openvla/openvla #3: Missing data needed to reproduce on other robots

nbubis, 2024-06-18:
In order to test this on a real robot one would need details on:

- The units of the resultant action (I'd assume meters? radians in angle-axis?)
- The coordinates used (Z = camera forward?)
- Grippers supported: I'd assume part of the image will always include gripper fingers, is the model agnostic to these somehow? How are grippers calibrated so that the action output can be applied to other grippers?
- Transformations from action coordinates to robot coordinates 

Cheers!

Kashu7100, 2024-06-27:
I have the same question.

moojink, 2024-06-28:
Hi @nbubis,

Thank you for the questions! Please find my answers below:

> The units of the resultant action (I'd assume meters? radians in angle-axis?)

The action space consists of 7 dimensions: The first 3 correspond to relative changes in the end-effector's XYZ position (continuous), the next 3 correspond to relative changes in the end-effector's orientation (continuous), and the last one corresponds to the opening/closing of the gripper (discrete: +1 = open, 0 = close). The exact units/representations of the actions are defined by each individual dataset, so you should consult the prior works for that. For example, for the DROID dataset, the relative changes in end-effector orientation are expressed in delta xyz Euler angles.

> The coordinates used (Z = camera forward?)

This also depends on how each individual dataset has defined its action space. For example, in BridgeData V2, `dX` = end-effector forward/backward, `dY` = end-effector left/right, `dZ` = end-effector up/down. This is a common convention, though there may be some datasets that do not follow it exactly. We do not manually align the coordinate frames of all the robots.

> Grippers supported: I'd assume part of the image will always include gripper fingers, is the model agnostic to these somehow? How are grippers calibrated so that the action output can be applied to other grippers?

Indeed, since we train on static third-person images of the robot scenes, the images includes the grippers. I wouldn't say that the model is truly "agnostic" to the gripper type, i.e., if you swap out the gripper in a seen environment with a different unseen gripper at test time, there is no guarantee that things will work very well. However, we hope that the model can generalize to some degree as long as you use the same action space. For example, if you collect expert demonstrations with the Franka Panda robot using the default gripper, and then swap out the gripper with a Robotiq gripper while keeping everything else fixed (including the action space, which means, e.g., binarizing continuous gripper actions to discrete actions), then it is possible that the trained policy could still work. Note that we only tried parallel-jaw grippers with binary open/close actions, so other types of grippers wouldn't work out of the box.

Also, note that we do not do any explicit calibration of actions or coordinate frames across environments (other than aligning the open/close gripper action), and we do not condition action generation on anything other than a single input image and language instruction. The OpenVLA model is likely automatically learning the necessary transformations to map action representations to the correct coordinate frame, based on the image it sees during generation. For example, if it sees a WidowX robot in an environment seen in the BridgeData V2 dataset, it knows how to represent "move forward and down" with the correct action tokens since it has learned this mapping during training.

> Transformations from action coordinates to robot coordinates

The transformations for the end-effector's coordinates are handled by each individual robot environment. For example, in DROID, the actions (which represent relative changes in pose) are applied to the current end-effector's pose to get the new pose, and then the inverse kinematics solver figures out the joint positions needed for the end-effector to reach that pose. The same idea applies in BridgeData V2.

Hope this helps!
-Moo Jin

closed 2024-06-30 by moojink

ssaroha, 2025-01-25:
Hi @moojink 
I came across this issue as I was searching for way to convert openVLA's end-effector output (Δx, Δy, Δz, Δroll, Δpitch, Δyaw, gripper) to lerobot's [so100 arm](https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md) that outputs joint spaces (angles of 6 joint motors in radians)

Your suggestion below 
> in DROID, the actions (which represent relative changes in pose) are applied to the current end-effector's pose to get the new pose, and then the inverse kinematics solver figures out the joint positions needed for the end-effector to reach that pose. The same idea applies in BridgeData V2.

means, I need to use inverse kinematics solver to figure out the 6 joint positions. What is the best way to do that? would highly appreciate any code pointers on how to do it.

My end goal is to get openVLA model working with lerobot so100 arm without wholesale retraining. I have also received suggestions to drop the last layer of openVLA and add a linear layer that outputs 6 values to control in joint space. but i think that will require retraining.

Any inputs on best way to go about it highly appreciated.

Thanks!