openvla/openvla #4: memory leak

yueyang130, 2024-06-18:
Hi, guys. Thanks for your great open-source work. When I use your rlds dataset built on tf.data.Dataset, I found there is memory leak when `train`=True. Specifically, when it's true and `shuffle` is called without `cache`, the memory gradually increases as the iteration goes.

https://github.com/openvla/openvla/blob/388244e88555150b67520419932189f459ad74cf/prismatic/vla/datasets/rlds/dataset.py#L572

However, I don't understand why it happens. In my understanding, `shuffle` preloads a fixed size of data before iteration and then replace the used data with new data on the fly during iteration. So the memory should keep constant after preloading.
Could you provide some solution to the memory leak? If the leak is hard to solve, could you provide an approximate memory it needs to run the whole iteration during training?

moojink, 2024-06-19:
Hi @yueyang130,

Thank you for the question. **Technically, this is not a "memory leak" -- i.e., it is not a bug.** The TFDS dataloader may use more memory than expected (likely due to aggressive prefetching or other operations happening under the hood in TFDS), and the memory utilization may look like it will increase forever, but it will actually stop increasing and stay constant at some point. If your system has limited RAM and the training shuffle buffer size is too large, your program may crash before you can observe that the memory utilization eventually plateaus.

**Therefore, if you are running into out-of-memory issues while training, we recommend that you decrease the value of `shuffle_buffer_size` in your VLA training configuration (see [`openvla/prismatic/conf/vla.py`](https://github.com/openvla/openvla/blob/main/prismatic/conf/vla.py)).** Every user has a different amount of RAM available, so we cannot recommend a default shuffle buffer size that will work for everyone. But just for reference, in some of our smaller-scale experiments during development, we had 1 TB of RAM available on a single node with 8 A100 GPUs, and we set `shuffle_buffer_size` to `256_000` (so our effective shuffle buffer size was 256K * 8 == ~2M samples, since we had spawned 8 processes that each had their own shuffle buffer; this is how our dataloader works by default). And this configuration would only be at ~70% system memory utilization even after 100K steps over 4-5 days, which was sufficient for our purposes. For your use case, you can experiment with the shuffle buffer size and try to reduce it while making sure that model training is still stable.

Once you decrease the `shuffle_buffer_size` to something that works for your machine, you will be able to confirm that there is no "memory leak" because the utilization will stabilize after a certain point.

Thank you for the question and hope this helps!
-Moo Jin

yueyang130, 2024-06-20:
Thanks for your kind response. I will try your advice later.

Another question is that your dataloader for vla training don't use DistributedSampler (since IterableDataset doesn't allow it...). However, will it result that each torch process have the same data instead of the situation that each process have s separate subset of the dataset?

yueyang130, 2024-06-20:
> Hi @yueyang130,
>
> Thank you for the question. **Technically, this is not a "memory leak" -- i.e., it is not a bug.** The TFDS dataloader may use more memory than expected (likely due to aggressive [...]

You're right! I use a machine with larger RAM. 8 processes with shuffle_size=256,000 finally utilizes 870G memory and stop increasing. Btw, could you please provide some explanation / experimental data to illusrate why shuffle is so important?

Toradus, 2024-06-21:
> Hi @yueyang130,
> 
> **Therefore, if you are running into out-of-memory issues while training, we recommend that you decrease the value of `shuffle_buffer_size` in your VLA training configuration (see [`openvla/prismatic/conf/vla.py`](https://github.com/openvla/openvla/blob/main/prismatic/conf/vla.py)).** Every user has a different amount of RAM available, so we cannot recommend a default shuffle buffer size that will work for everyone. But just for reference, in some of our smaller-scale experiments during development, we had 1 TB of RAM available on a single node with 8 A100 GPUs, and we set `shuffle_buffer_size` to `256_000` (so our effective shuffle buffer size was 256K * 8 == ~2M samples, since we had spawned 8 processes that each had their own shuffle buffer; this is how our dataloader works by default). And this configuration would only be at ~70% system memory utilization even after 100K steps over 4-5 days, which was sufficient for our purposes. For your use case, you can experiment with the shuffle buffer size and try to reduce it while making sure that model training is still stable.

One question regarding this part: I currently use a very similar setup (same Dataloader, because it is a very smooth way to load the OXE Dataset) but with HF Accelerate instead of directly using torch.nn.parallel.DDP.
Why did you chose to initialize 1 Dataloader per prozess and not share the memory?

My training speed is currently a little bit low but im not sure if i configurated accelerate wrong or if it is coming from the Dataloader. From your code i can see that you experimented with it a little [run_training function in base strategy](https://github.com/openvla/openvla/blob/ca9ae9b6014eb1568c36dfc641b90a9311abaaf8/prismatic/training/strategies/base_strategy.py#L106) with workers = 2. Are there some findings you can report?
Im currently doing my Master Thesis and still struggle with it, so it would be really nice to get some more insights :)

moojink, 2024-06-21:
Hi @yueyang130,

> Thanks for your kind response. I will try your advice later.
> 
> Another question is that your dataloader for vla training don't use DistributedSampler (since IterableDataset doesn't allow it...). However, will it result in that each torch process have the same data instead of the situation that each process have s separate subset of the dataset?

You're right that `DistributedSampler` doesn't work well with `IterableDataset` by default. However, each process will not have the same sequence of data samples at training time. This is because when the dataset is initialized at training time, the sharded dataset files (TFRecords) are shuffled before they are loaded. Therefore, assuming that the dataset is sharded into multiple TFRecords and not just a single TFRecord, each process will have a dataloader with a different ordering of training samples. You can verify this by initializing the dataloader for a large dataset -- multiple times across multiple runs -- and observing that the first few samples differ every time. (Note, however, that samples _within_ TFRecords are not shuffled during dataset initialization, so we still need to call `dataset.shuffle(shuffle_buffer_size)` to ensure that local shuffling is being done.)

> You're right! I use a machine with larger RAM. 8 processes with shuffle_size=256,000 finally utilizes 870G memory and stop increasing. Btw, could you please provide some explanation / experimental data to illustrate why shuffle is so important?

Great to hear that! Regarding the importance of shuffling: We haven't conducted extensive analysis of the shuffle buffer size and its relation to policy performance for this project due to compute constraints. In the beginning, we simply chose to use a large shuffle buffer -- as much as we could fit in RAM: ~2M samples for single-node training with 8 GPUs, and ~16M for multi-node training with 8 nodes x 8 GPUs each. We did this because the [Octo paper](https://arxiv.org/pdf/2405.12213) mentions that using a larger buffer (e.g., 500K samples) was important for improving policy performance. Also, intuitively, we want training samples to be as independent and identically distributed as possible.

moojink, 2024-06-21:
Hi @Toradus,

> One question regarding this part: I currently use a very similar setup (same Dataloader, because it is a very smooth way to load the OXE Dataset) but with HF Accelerate instead of directly using torch.nn.parallel.DDP. Why did you choose to initialize 1 Dataloader per prozess and not share the memory?

We opted for the simplest implementation based on native PyTorch Distributed training logic, which automatically spawns one dataloader per process by default if you use `torchrun`. It is possible to initialize a single dataloader that is shared across processes (the Hugging Face Trainer library may be doinig this already, for example), but we found that this was not necessary to implement given that we observed good training performance without it.

> My training speed is currently a little bit low but im not sure if i configurated accelerate wrong or if it is coming from the Dataloader. From your code i can see that you experimented with it a [...]

That default number of workers set to `2` was just a default value that had worked well during our development, but it may be system-dependent, so I would recommend that you experiment with a different value if you find that dataloading is the bottleneck in your training speed. However, we observed that dataloading speed is usually not the bottleneck; dataloading is usually much faster than operations with these large models, such as computing gradients and taking gradient steps. I would recommend that you profile different parts of your training script to see what the current bottleneck is before experimenting further with the dataloader!

yueyang130, 2024-06-21:
Hi @moojink,

  Thanks a lot for your explanation!

closed 2024-06-21 by yueyang130

zhihou7, 2024-07-07:
Hi, I observed the memory keeps increasing util crashed.  The memory first quickly increases, and then keeps slowly increasing. Do you have any suggestions?

[image]

LukeLIN-web, 2025-03-21:
> Hi, I observed the memory keeps increasing util crashed. The memory first quickly increases, and then keeps slowly increasing. Do you have any suggestions?
> 
> <img alt="image" width="1222" src="https://private-user-images.githubusercontent.com/67958240/346338055-c143d7ee-a576-4f6c-9a2a-67c9f94197b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI1MjY0OTEsIm5iZiI6MTc0MjUyNjE5MSwicGF0aCI6Ii82Nzk1ODI0MC8zNDYzMzgwNTUtYzE0M2Q3ZWUtYTU3Ni00ZjZjLTlhMmEtNjdjOWY5NDE5N2I1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMzIxVDAzMDMxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ5ZmNjMmFlOTM0MTYzY2M5NmQyZjE2Y2ViODAxMGJiNDlhMTQ0MGY4Yjk3N2YyYjZhZjBiNmQ4YmM5NDdkMDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PtJQ095ifQOmzpzQwjY-3vCd1dFFXlytpgGZy6xhKlw">
> By the way, does the dataloader is necessary. Could we use the batch from tf.dataset?

1. decrease `shuffle_buffer_size`
2.  modify the dataloader to not copy the dataset
3.

LukeLIN-web, 2025-04-30:
lisaopenvla/prismatic/vla/datasets/datasets.py

```
   # Validation =>> fix a single shuffle buffer of data and cache it in RAM; prevents gradual memory increase!
    if not train:
        dataset = dataset.take(shuffle_buffer_size).cache()

    # Shuffle the Dataset
    #   =>> IMPORTANT :: Shuffle AFTER .cache(), or else memory will still leak!
    dataset = dataset.shuffle(shuffle_buffer_size)

```

I wonder whether if we use `.cache() ` when training can solve this problem

mikygit, 2025-05-21:
> I wonder whether if we use `.cache() ` when training can solve this problem

Goof question.
Are there any reasons or drawbacks not to do it?
Because it seems to prevent the OOM issue ...

LukeLIN-web, 2025-11-11:
> I meet same problem how to solve?

decrease shuffle_buffer_size