openvla/openvla #5: Suggestion to add extern script imports to separate requirements file and request for Figure 3,4 images

vrtnis, 2024-06-20:
Hi all,

Thank you for this fantastic project and repository! I'm currently working with the REPL example to create an OpenVLA demo/walkthrough and wanted to suggest a couple of improvements:

Separate Requirements File: It would be helpful to add some of the imports, such as draccus, dlimp, and rich, to a separate requirements-extern.txt file within the extern/REPL example folder. This could streamline the setup process for users.

Higher Resolution Images:  Could you please provide higher resolution versions of the images (without annotations) from Figure 3 and Figure 4 of the paper? Perhaps these could be included in the extern folder if possible.

Thanks again for all your hard work on this project!

siddk, 2024-06-20:
Hey @vrtnis - is the top-level `requirements-min.txt` file not sufficient? Or is it missing some imports - I thought I had scoped it to include everything necessary for the README example, but happy to update it if I was wrong!

(I don't want to separate the repo requirements like `dlimp`, `draccus`, or `tensorflow-datasets` from the pyproject.toml, just because they are necessary for running the training and full fine-tuning scripts.)

vrtnis, 2024-06-20:
Hey @siddk , thanks for your quick response!  Yep, agree with not separating the repo requirements.  I think `requirements-min.txt` might be missing some imports but here is some more context.

I did run `pip install -r https://raw.githubusercontent.com/openvla/openvla/main/requirements-min.txt`, however, after that, when I did run `generate.py` in the `external` folder, I started seeing module not found errors initially for `dlimp`, `draccus`, `rich` and `tensorflow_graphics`.  

Here is one example below and the ones for other modules were similar. 

 I'm not sure if its just me, but I  can imagine that if all these modules were included in the `requirements-min.txt` these errors probably wouldn't occur.

`2024-06-20 18:56:16.446780: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-06-20 18:56:16.446832: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-06-20 18:56:16.447826: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-06-20 18:56:16.452965: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-06-20 18:56:17.177750: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
/usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.4
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
Traceback (most recent call last):
  File "/ephemeral/users/vrtnis/openvla/openvla_repl1.py", line 9, in <module>
    from prismatic import load
  File "/ephemeral/users/vrtnis/openvla/prismatic/__init__.py", line 1, in <module>
    from .models import available_model_names, available_models, get_model_description, load
  File "/ephemeral/users/vrtnis/openvla/prismatic/models/__init__.py", line 1, in <module>
    from .load import available_model_names, available_models, get_model_description, load, load_vla
  File "/ephemeral/users/vrtnis/openvla/prismatic/models/load.py", line 18, in <module>
    from prismatic.models.vlas import OpenVLA
  File "/ephemeral/users/vrtnis/openvla/prismatic/models/vlas/__init__.py", line 1, in <module>
    from .openvla import OpenVLA
  File "/ephemeral/users/vrtnis/openvla/prismatic/models/vlas/openvla.py", line 17, in <module>
    from prismatic.vla.action_tokenizer import ActionTokenizer
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/__init__.py", line 1, in <module>
    from .materialize import get_vla_dataset_and_collator
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/materialize.py", line 18, in <module>
    from prismatic.vla.datasets import EpisodicRLDSDataset, RLDSBatchTransform, RLDSDataset
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/__init__.py", line 1, in <module>
    from .datasets import DummyDataset, EpisodicRLDSDataset, RLDSBatchTransform, RLDSDataset
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/datasets.py", line 23, in <module>
    from prismatic.vla.datasets.rlds.oxe import OXE_NAMED_MIXTURES, get_oxe_dataset_kwargs_and_weights
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/rlds/oxe/__init__.py", line 1, in <module>
    from .materialize import get_oxe_dataset_kwargs_and_weights
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/rlds/oxe/materialize.py", line 13, in <module>
    from prismatic.vla.datasets.rlds.oxe.configs import OXE_DATASET_CONFIGS, ActionEncoding
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/rlds/oxe/configs.py", line 29, in <module>
    from prismatic.vla.datasets.rlds.oxe.utils.droid_utils import zero_action_filter
  File "/ephemeral/users/vrtnis/openvla/prismatic/vla/datasets/rlds/oxe/utils/droid_utils.py", line 6, in <module>
    import tensorflow_graphics.geometry.transformation as tfg
ModuleNotFoundError: No module named 'tensorflow_graphics'`

siddk, 2024-06-20:
Ahh - ok, so this is our bad. The scripts in `scripts` (e.g., `generate.py`) are all holdovers from the original repo at [TRI-ML/prismatic-vlms](https://github.com/TRI-ML/prismatic-vlms) -- these support the base visually-conditioned language models we trained OpenVLA from, but not the OpenVLA models themselves -- we trained the OpenVLA models only to generate robot actions, so they (unfortunately) do not generate meaningful natural language right now.

Two options here - let me know what you think. We could either get rid of the `scripts` directory in this fork to avoid confusion, or I could drop a note in the README that `vla-scripts` is meant for running and loading OpenVLA models.

(One benefit of leaving `scripts` in this directory is that you *can* train VLMs still in this repo, but at the cost of all the extra dependencies...)

vrtnis, 2024-06-21:
Ah that totally makes sense - thanks for the explanation! (and also explains the natural language output by generate.py REPL but _not_ the widow README example).  
Despite this, I think it does make sense to leave it in with the note that you suggest.  Removing it would probably create more questions as to why `scripts` not there (despite there being a good reason).

closed 2024-06-21 by siddk