openvla/openvla #9: The issue about bridge dataset

Zhangwenyao1, 2024-06-27:
Thanks for your great work！
I want to know choose which version of bridge dataset to train Openvla？
If it is bridge v2，the website of you provide is version 1.

siddk, 2024-06-28:
Hey @Zhangwenyao1 -- sorry for the confusion; we use [Bridge **V2**](https://rail-berkeley.github.io/bridgedata/).

The [link we provided in the README](https://rail.eecs.berkeley.edu/datasets/bridge_release/data/tfds/bridge_dataset/) resolves to the same place as the "Data" link on the Bridge V2 website.. unfortunately, just because of RLDS convention, it is labeled as 1.0.0 (**but it is a different dataset than the original Bridge V1 data linked from OXE**).

closed 2024-06-28 by siddk

Zhangwenyao1, 2024-06-29:
Thanks for your reply. 
It sitll confusing because the parent directory of this link has a zip file named “demos_8_17.zip” arround 411G(
[demos_8_17.zip](https://rail.eecs.berkeley.edu/datasets/bridge_release/data/demos_8_17.zip))， and the children directory of your link named “1.0.0” included subfiles like “bridge_dataset-train.zip” with totally 100G. (https://rail.eecs.berkeley.edu/datasets/bridge_release/data/tfds/bridge_dataset/1.0.0/bridge_dataset-train.tfrecord-00000-of-01024)https://rail.eecs.berkeley.edu/datasets/bridge_release/data/tfds/bridge_dataset/）. 
So I will appreciate it if you can tell me which one should i choose.

LukeLIN-web, 2025-02-16:
> Thanks for your reply. It sitll confusing because the parent directory of this link has a zip file named “demos_8_17.zip” arround 411G( [...]

I download demos_8_17.zip, it seems different with  “1.0.0”  structure 

```
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_22.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_25.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_10.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_17.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_19.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_6.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-09/raw/traj_group0/traj45/images0/im_1.jpg  
  inflating: bridge/raw/rss/toykitchen7/pnp_sweep_target_fixed/00/2023-03-01_12-50-
```