openvla/openvla #17: How does it work on a novel image ?

jis3613, 2024-07-03:
Hello, first of all, I want to thank the authors for this great work they have done and providing the whole work as an open-source. 
It truly contributes to the robot learning research and boosts the development of general robot intelligence.

I have some questions about how the trained model, "openvla-7b" outputs the action (7 Dof dimensions) for a novel input image.

With the code you provided in the Getting Started in README, I could feed a novel image I personally have to the OpenVLA along with an instruction prompt.

And it successfully outputs the 7 action dimensions. (I did not validate the accuracy of the action output yet though.)

The given image shows a picture of a table with multiple objects on it and a robotic arm.
It is an image I personally took so the OpenVLA could not have seen it before during its training.
Therefore, it does not know the camera position, the robot position and anything else in the image.
Only data given to OpenVLA is **the image** and the **instruction prompt**.

Then, without these information, how does OpenVLA accurately calculate the 7 action dimensions ?

Does it estimate the positions of the view-point of camera, robot and objects only with the given image?

I read your answers on other [issues]( #3, #6 ) saying that

"we do not do any explicit calibration of actions or coordinate frames across environments (other than aligning the open/close gripper action), and we do not condition action generation on anything other than a single input image and language instruction. 
The OpenVLA model is likely automatically learning the necessary transformations to map action representations to the correct coordinate frame, based on the image it sees during generation. "

and 

"If you want to use the model for a certain target configuration, you can just finetune it with a bit of data from your target domain and it will adopt the action space you need and learn to output such actions. If you want to use the model out of the box (ie without finetuning), you'd need to match one of the setups from the training datasets pretty closely (eg set up a Bridge-like WidowX with the BridgeDataV2 control stack) and then the model would output actions for the action space definition used in that dataset (implicitly, by recognizing the robot in the images)."

However, I still do not quite understand how it outputs actions for a novel image input. 

Does it use the coordinate systems it learned from its training data? such as BridgeData V2?

Then, would the action output from my input image can be quite inaccurate ?

Also, does the parameter "unnorm_key" in predict_action function refer to the dataset it uses to un-normalize the positions and actions spaces ?

Also, I am wondering whether the 7 dimension action output is the final position of the robot or just a one piece of the trajectory to the final position.

Does it output a trajectory of the action for each time step to get to the final goal position?

If it is latter, how frequently does it output the 7 action dimensions for a single instruction output?

Thanks,

kpertsch, 2024-07-04:
Thanks for the questions!
The model will produce 7-dim actions for any image you feed to it, but indeed unless these images come from a setup that is very similar to the training setup (eg familiar camera angle, robot, scene etc) it's unlikely the out-of-the-box model will produce accurate actions, since as you mentioned the model does not have any information about the coordinate system etc that's used in your scene. For the model to work well in a new scene you would need to finetune it with a bit of data from your scene.

The unnorm_key refers to the dataset statistics that are used to unnormalize the predicted actions. During training, we normalize the actions from each training datasets with min/max stats computed *per dataset*. So the unnorm key determines which stats are used at test time. If you finetuned the model, you should use the stats from your finetuning dataset.

The 7-dim action is a delta end-effector action, ie it corresponds to 3d translation, 3d rotation and a 1d gripper *position* value. Check the paper for achievable inference speed (and higher speed is possible with further optimization). The model will predict actions that match the control frequency of the training / finetuning data.

jis3613, 2024-07-04:
Thanks for the answer. I think things are a bit more clear now.

closed 2024-07-08 by siddk

peihaochenumass, 2024-07-23:
Thanks for your great work! @kpertsch 

As you mentioned above, "the model will produce 7-dim actions for any image you feed to it, but indeed unless these images come from a setup that is very similar to the training setup (eg familiar camera angle, robot, scene etc) it's unlikely the out-of-the-box model will produce accurate actions".

Then, my question is: how would you perform out-of-the-box on the WidowX robot from the Bridge V2 dataset (as described in Sec 5.1 in the paper)? More specifically, how would you set the camera to the same pose as the Bridge V2 training data during testing?  I don't think the model can perform well if the camera pose is different during testing and training. 

Many thanks.

kpertsch, 2024-07-23:
That's correct, so you need to carefully position the camera to be well-aligned with training scenarios. We usually found it helpful to code up a little visualizer that overlays an image from the bridge dataset you're trying to match with the current camera live stream which makes it easier to find the right angle. After you think you have a good angle you can just run OpenVLA to see whether it performs reasonable movements to check whether your angle is right.

peihaochenumass, 2024-07-24:
Thanks for your quick reply! I will try it later.

peihaochenumass, 2024-07-24:
Actually, I am trying to apply the trained OpenVLA model to the Franka robot in our lab and have the following questions:

1. Since different training data in Open-X have different robot settings (eg, the definition of XYZ, the position of the robot base), how does the model decide which robot setting should be followed when the model is applied on a new scene? I would expect the model to choose the robot setting according to the observed robot arm, the testing scene, and the camera pose. It may match the most similar training dataset to get the corresponding robot setting and then output the actions according to the action space in that robot setting. Am I right?

2. The robot in my lab is Franka. If I test this robot, which robot setting would the model follow? I saw several training datasets in Open-X that have used Franka. Among these datasets, DROID dominates. Should I set the camera pose similar to DROID dataset to get the best "out-of-the-box" performance? According to DROID paper, they capture 3rd person's view using two randomly set cameras. It seems that it's hard for me to determine my camera pose to keep the same as DROID. Can you give us any suggestions?

Thanks!

jis3613, 2024-08-01:
I am also curious about the above questions.

tongvibe, 2024-08-08:
> Actually, I am trying to apply the trained OpenVLA model to the Franka robot in our lab and have the following questions:
>
> 1. Since different training data in Open-X have different robot settings [...]

I have been using octo and openvla for a while, and I've done some fine-tuning on my own dataset with octo. Your first and second questions can be answered together. My overall impression is that the model outputs a delta action—a difference in x, y, z, roll, pitch, and yaw. This means the model essentially predicts how the end-effector should move in space. So in other words, the model doesn't really care what specific robotic arm you're using; it just needs your training set to include some general definitions. It doesn't need to explicitly know the exact scenario or which robotic arm is used since the critical parameters involved in the training are just your action and images. The robotic arm serves mainly as a background, second, to normalize and denormalize the input action, and third, to provide a correspondence between the robotic arm and the action (which means in some training sets, the robotic arm might not even be visible in the main scene and the model just fits to a similar robotic arm based on the similar scenario). This is more of a dataset issue. This is also why the author emphasizes that you either need a setup that is highly similar to the experimental setup or you need to fine-tune for your specific scenario. 

My experience tells me that it is almost impossible to set up an extremely similar scene due to the reasons mentioned in your first question: you can't perfectly determine the robot's base, any translations, the camera angle, whether you're using a rotation matrix or Euler angles, among other factors. Additionally, my experiments have shown that the camera angle is very important, and a small number of datasets (e.g., 10 sets) have almost no effect on the model.

These are just some of my personal insights; please forgive any errors. For concrete answers, we should refer to the author's response.

For my personal plan, I aim to have at least 100 sets in the custom model as a general definition, which can be understood as telling the model that we have a new scenario and wanting it to understand it first. After that, I would proceed with small group training of the model. For example, testing its generalization with 10 sets of camera angle adjustments, 10 sets of different tasks in the same scene, and so on. To illustrate, it's like a person entering a new scene and needing to adapt and understand the environment first.

---