openvla/openvla #18: get_openvla_prompt ablations

willxxy, 2024-07-03:
Hello,

Thank you for the great works. I had a question out of curiosity if there has been any ablation studies you did in regards to the system prompt in `openvla/vla-scripts/deploy.py` lines 58 - 62. From my understanding, it seems only this prompt of f"{SYSTEM_PROMPT} USER: What action should the robot take to {instruction.lower()}? ASSISTANT:" or f"In: What action should the robot take to {instruction.lower()}?\nOut:" were tested. 

If you could expand on the sensitivity of the model's performance wrt the system prompt it would be great!

Thank you.

willxxy, 2024-07-09:
Another update to this issue. For questions 1 and 2, from my experience, it seems like for inference, f"{SYSTEM_PROMPT} USER: What action should the robot take to {instruction.lower()}? ASSISTANT:" or f"In: What action should the robot take to {instruction.lower()}?\nOut:" does not really give different performance. Is this a normal behavior?

moojink, 2024-07-10:
Hi @willxxy,

Thank you for reaching out. Here are answers to your questions:

> I had a question out of curiosity if there has been any ablation studies you did in regards to the system prompt in openvla/vla-scripts/deploy.py lines 58 - 62.

1. We have not tried altering the system prompt in our experiments. We simply chose to use the same system prompt used by the base Prismatic VLM. The reason for this is that if the VLM has always seen the same system prompt during the pretraining phase, then changing the system prompt during fine-tuning or inference would introduce distribution shift with respect to the training data, which would not be ideal.

Note: The system prompts for the main OpenVLA model ([openvla-7b](https://huggingface.co/openvla/openvla-7b)) and the early development model ([openvla-v01-7b](https://huggingface.co/openvla/openvla-v01-7b)) differ. If using the main model, you should use the prompt structure `f"In: What action should the robot take to {instruction.lower()}?\nOut:"`. If using the early development model, you should use `f"{SYSTEM_PROMPT} USER: What action should the robot take to {instruction.lower()}? ASSISTANT:"`, where `SYSTEM_PROMPT == "A chat between a curious user and an artificial intelligence assistant. The assistant gives helpful, detailed, and polite answers to the user's questions."` These are the system prompts used by the respective base VLMs.

> Additionally, in the Getting Started code snippet in the README page
> I was wondering why the prompt did not have the SYSTEM_PROMPT?

2. The "Note" in my answer above addresses this. The [openvla-7b](https://huggingface.co/openvla/openvla-7b) model was not pretrained with that `SYSTEM_PROMPT` (including the base Prismatic VLM), so you should omit it when running inference with that model.

> Lastly, could you please clarify whether the given logic is correct for deployment?

3. If using the [openvla-7b](https://huggingface.co/openvla/openvla-7b) model, you would need to change the system prompt to match the Getting Started code snippet. Aside from that, the overall structure of your code snippet looks right to me!

> Another update to this issue. For questions 1 and 2, from my experience, it seems like for inference, f"{SYSTEM_PROMPT} USER: What action should the robot take to {instruction.lower()}? ASSISTANT:" [...]

That is surprising to me, given that you are changing the tokens that are being fed into the model. Perhaps you observed similar performance for some narrow dataset/example, but I would not expect this to hold in general. I would recommend that you use the correct system prompt to avoid distribution shift.

Hope that helps!
-Moo Jin

closed 2024-07-10 by moojink