openvla/openvla #20: Pre-processing the BridgeData V2 in RLDS format

jis3613, 2024-07-05:
Hi, me again!

I'm trying to train the OpenVLA model on BridgeData V2 with the script you provided in the README.

```bash
# Train VLA on BridgeData V2 with the Prismatic DINO-SigLIP 224px Backbone on a Single Node (w/ 8 GPUs)
torchrun --standalone --nnodes 1 --nproc-per-node 8 vla-scripts/train.py \
  --vla.type "prism-dinosiglip-224px+mx-bridge" \
  --data_root_dir <PATH TO OXE DATA ROOT> \
  --run_root_dir <PATH TO LOG/CHECKPOINT ROOT> \
  --wandb_project "<PROJECT>" \
  --wandb_entity "<ENTITY>"
```

I downloaded the  BridgeData V2 from its official website and placed it under `openvla/datasets/open-x-embodiment/`.

I tried to pre-process the dowloaded dataset with the [custom script](https://github.com/moojink/rlds_dataset_mod/blob/main/prepare_open_x.sh) mentioned in README.
This is the command I ran.

```bash
DOWNLOAD_DIR="../datasets/open-x-embodiment"
CONVERSION_DIR="../datasets/temp"
N_WORKERS=20                  # number of workers used for parallel conversion --> adjust based on available RAM
MAX_EPISODES_IN_MEMORY=200    # number of episodes converted in parallel --> adjust based on available RAM

# increase limit on number of files opened in parallel to 20k --> conversion opens up to 1k temporary files
# in /tmp to store dataset during conversion
ulimit -n 20000

DATASET="bridge"
TRANSFORM="resize_and_jpeg_encode"

python3 modify_rlds_dataset.py  --dataset=$DATASET --data_dir=$DOWNLOAD_DIR --target_dir=$CONVERSION_DIR --mods=$TRANSFORM --n_workers=$N_WORKERS --max_episodes_in_memory=$MAX_EPISODES_IN_MEMORY
```

`modify_rlds_dataset.py` is located at `openvla/rlds_dataset_mod/` .

When I run the above command, I get the following error message.

```bash
File "~/openvla/rlds_dataset_mod/rlds_dataset_mod/multithreaded_adhoc_tfds_builder.py", line 149, in __init__
   super().__init__(*args, **kwargs)
TypeError: SplitBuilder.__init__() got an unexpected keyword argument 'file_format'.
```

Do you know any way to resolve this error?
Also, I'm wondering if it is necessary to pre-process the downloaded BridgeData V2,
because when I run `modify_rlds_dataset.py`, It seems that it tries to download the dataset from the server again.

Thanks!

moojink, 2024-07-10:
Hi @jis3613,

Thank you for reaching out. We recently pushed new sections to the README with more detailed instructions: [Fine-Tuning OpenVLA via LoRA](https://github.com/openvla/openvla?tab=readme-ov-file#fine-tuning-openvla-via-lora) and [Fully Fine-Tuning OpenVLA](https://github.com/openvla/openvla?tab=readme-ov-file#fully-fine-tuning-openvla). You can find our recommended commands for downloading the BridgeData V2 dataset specifically, in both of those sections.

Please try that out and see if it works for you. I'll close this issue for now, but feel free to reopen it if the issue is not resolved.

Hope that helps!
-Moo Jin

closed 2024-07-10 by moojink

jis3613, 2024-07-11:
Awesome! Thanks for the update!