openvla/openvla #21: Un-normalizing statistics for Google robot tasks

HuFY-dev, 2024-07-06:
Hi, I'm playing with the OpenVLA model and I want to evaluate this model on [SimplerEnv](https://github.com/simpler-env/SimplerEnv/tree/main)'s Google robot tasks. Since there is a `unnorm_key` argument in the `predict_action` method, I assume that that's related to some dataset-specific bias terms and has to be changed if I use some other datasets. However I saw that for the [openvla-7b](https://huggingface.co/openvla/openvla-7b) model, there's no google-related keys in the `norm_stats` dictionary. Does that mean I have to retrain/fine-tune the model on Google tasks? Also, I saw you did some Google robot evaluations in the paper, have you released related code? Thanks in advance for your help!

HuFY-dev, 2024-07-07:
Never mind, I think I found the `unnorm_key` for Google which is `fractal20220817_data`. I also want to make sure that I should use `bridge_orig` for SimplerEnv's WidowX tasks right?

HuFY-dev, 2024-07-07:
I'm struggling to get OpenVLA to work on this kind of scenarios:

Task: pick up spoon and place on towel

[image]

I was using the `bridge_orig` as the unnorm key, but that might be not the correct option. Do you know which key I should choose?

HuFY-dev, 2024-07-08:
Hi @siddk @kpertsch @moojink, sorry to ping you guys, but do you have any clues what unnorm key I should use in the above scenario? Thanks in advance for the help!

kpertsch, 2024-07-09:
Hey sorry for the late reply!
The `unnorm_keys` you are using are correct for both Bridge and Google Robot.
When I tried SIMPLER before, OpenVLA seemed to work pretty well on the Google robot tasks and less well on the Bridge tasks, since there the visual real-to-sim domain gap is wider. Can you try the google robot tasks?

I am pasting my SIMPLER script below -- it won't quite run out of the box since it's not yet adapted to the release code (we need a bit of time to clean this up), but figured it could be helpful for you to see the sticky action logic required for the Google Robot (this is copied from SIMPLER Octo eval, which used the same logic).

The reason Octo & OpenVLA need sticky gripper is that during their training we convert the "relative gripper actions" -- +1 for opening, -1 for closing -- into "absolute gripper actions" -- +1 for *opened*, -1 for *closed*. So during inference we need to convert them back to relative, but need to apply the relative opening / closing actions for multiple timesteps in a row for the gripper to actually fully close.

Hope this helps your investigations!

```
"""
eval_model_in_simpler_env.py

Runs a model checkpoint in a simulated SIMPLER environment.

Usage:
    python experiments/robot/libero/eval_model_in_simpler_env.py \
        --model.type <VLM_TYPE> \
        --pretrained_checkpoint <CHECKPOINT_PATH>
"""

import os
import sys
import time
from dataclasses import dataclass, field
from pathlib import Path
from typing import Union

import draccus
import numpy as np
import tqdm

import wandb
from prismatic.conf import ModelConfig, ModelRegistry

# TODO (moojink) Hack so that the interpreter can find experiments.robot
sys.path.append("../..")
from experiments.robot.simpler.simpler_utils import (
    get_simpler_env,
    get_simpler_img,
    process_simpler_action,
)
from experiments.robot.utils import (
    get_action,
    get_image_resize_size,
    get_model,
)

assert "MS2_ASSET_DIR" in os.environ, (
    "Environment variable MS2_ASSET_DIR not set. "
    "Usage: `MS2_ASSET_DIR=./ManiSkill2_real2sim/data python test_real2sim.py ...`"
)
DATE_TIME = time.strftime("%Y_%m_%d-%H_%M_%S")

@dataclass
class GenerateConfig:
    # fmt: off

    # Pre-trained model class
    model: ModelConfig = field(
        default_factory=ModelConfig.get_choice_class(ModelRegistry.REPRODUCTION_7B.model_id)
    )
    model_family: str = "llava"

    # Model Parameters
    pretrained_checkpoint: Union[str, Path] = Path(
        "/shared/karl/models/open_vla/siglip-224px+mx-oxe-magic-soup+n8+b32+x7/"
        "checkpoints/step-152500-epoch-27-loss=0.1637.pt"
    )
[... 106 lines cut ...]

        task_episodes += 1
        task_successes += float(done)
        print(f"Success rate: {float(task_successes) / float(task_episodes)}")
        if episode_idx < cfg.num_save_videos:
            # Save rollout GIF.
            wandb.log(
                {
                    f"video_{episode_idx}": wandb.Video(
                        np.array(rollout_images[:: cfg.video_temp_subsample]).transpose(0, 3, 1, 2)
                    )
                }
            )

    # Log and update total metrics
    wandb.log(
        {
            "success_rate": float(task_successes) / float(task_episodes),
            "num_episodes": task_episodes,
        }
    )

if __name__ == "__main__":
    eval_simpler()
```

closed 2024-07-09 by kpertsch

zhou-pig, 2024-08-29:
> Thank you!
Hello, my friend, I have the same problem as you. No matter what unnorm_key I choose, using OpenVLA doesn't work well in simpler env. Have you solved the problem?