openvla/openvla #23: About Hardware

yangwangopen, 2024-07-07:
How to purchase the corresponding hardware in the demonstration?

[image]
[image]
[image]

Or do you have any recommendations for other hardware to purchase?

kpertsch, 2024-07-07:
For the Bridge setup in the first picture, you can find a hardware setup guide here: https://docs.google.com/document/d/1si-6cTElTWTgflwcZRPfgHU7-UwfCUkEztkH3ge5CGc/edit

For a Franka setup comparable to the second picture, there's a hardware setup guide here: https://droid-dataset.github.io/droid/docs/hardware-setup 

The Google robot in the third image is not currently purchasable.

closed 2024-07-08 by siddk

Boltzmachine, 2025-05-23:
If I set up the image acquisition myself, how can I ensure that the image distribution remains consistent and is not affected by environmental factors like lighting? Or is this something people generally don't pay much attention to?