openvla/openvla #24: Dataset setup for finetuning

jis3613, 2024-07-08:
Hi, I'm trying to finetune the openvla-7b model on the bridge v2 dataset.
I know the model is already trained on bridge v2. 
I just want to have hands-on experience on finetuning the model.

I downloaded the bridge v2 dataset under openvla/datasets/open-x-embodiment/bridge_orig

and I run the following code
```bash
torchrun --standalone --nnodes 1 --nproc-per-node 8 vla-scripts/finetune.py \
                                 --data_root_dir "./datasets/open-x-embodiment/" \
                                 --dataset_name bridge \
```

Then I get the following error

```
Did you mean: bridge_orig -> bridge ?
The builder directory datasets/open-x-embodiment/bridge_orig doesn't contain any versions.
No builder could be found in the directory: datasets/open-x-embodiment for the builder: bridge_orig.
No registered data_dirs were found in:
     - datasets/open-x-embodiment
```

I would greatly appreciate it if you could help me on this issue.

youliangtan, 2024-07-08:
the dataset name should be: `--dataset_name bridge_orig`. And the directory should contain versioning, in this particular format: `<dataset_name>/0.1.0/xxx.tfrecord-0000X`

moojink, 2024-07-10:
Hi @jis3613,

Thank you for reaching out. We recently pushed new sections to the README with more detailed instructions: [Fine-Tuning OpenVLA via LoRA](https://github.com/openvla/openvla?tab=readme-ov-file#fine-tuning-openvla-via-lora) and [Fully Fine-Tuning OpenVLA](https://github.com/openvla/openvla?tab=readme-ov-file#fully-fine-tuning-openvla). You can find our recommended commands for downloading the BridgeData V2 dataset specifically, in both of those sections. Then you can follow the instructions to fine-tune OpenVLA (either with LoRA via Hugging Face `transformers`, or with full fine-tuning via PyTorch FSDP).

Please try that out and see if it works for you. I'll close this issue for now, but feel free to reopen it if the issue is not resolved.

Hope that helps!
-Moo Jin

closed 2024-07-10 by moojink

jis3613, 2024-07-11:
Great! Thanks!

lucywang720, 2024-09-01:
i met the same error, how do you solve it?

split600, 2024-10-12:
did you solve it . i met the same question