openvla/openvla #26: Publish Image Encoders as independent Models?

mbreuss, 2024-07-09:
Hi,
thank you for sharing this cool project and pretrained models! 
I'm particularly interested in the finetuned Siglip and Dino image encoders as standalone models.
I believe there could be a general interest in these finetuned image encoder models in the community. To my knowledge, there are no publicly available image encoders that have been finetuned on such a large dataset of 970,000 robot trajectories in PyTorch, especially with language supervision. Making these encoders easily accessible would be invaluable for researchers working on various robotic projects that may not require the full VLA model.
Would it be possible to release these finetuned image encoders as separate, standalone models? 
Thanks in advance!

siddk, 2024-07-09:
Hey @mbreuss -- yeah we definitely can; would you want those to just be parsed out of the existing checkpoint files? Any API you'd specifically like to see (right now it'd be easy to just get the image patch features before/after the projection layer -- anything else might require some work).

mbreuss, 2024-07-10:
Hi @siddk thanks for the fast response! I would be mainly interested in having access to the weights of SigLip and DinoV2 models separately before the projection layer, as both together are very compute intensive for smaller policies. If thats difficult the combined encoder before the projection is also fine.

kpertsch, 2024-07-12:
It would probably not be hard to release those as separate encoders, though we haven't tested whether they alone are any better than plain image-pre-trained encoders.

@mbreuss if you want to start experimenting with this, our encoders are simple timm modules, so it shouldn't be hard to instantiate the corresponding timm modules (you can check our code how they are instantiated) and then manually copy out the vision encoder weights from our checkpoints.
If you end up comparing to some existing visual representations for control, let us know how it goes :)

mbreuss, 2024-07-12:
@kpertsch thanks I plan to give it a try.  If I get any good results, I will let you know!

closed 2024-07-12 by mbreuss