openvla/openvla #27: finetuning with flexible observations and flexible action spaces

pxlong, 2024-07-09:
Thank you for open-sourcing the VLA model. This initiative will significantly foster the advancement of foundational models for robotics.

I'm curious whether you have plans to support fine-tuning with customizable/flexible observations (e.g. add proprio info, left/right wrist images, depth images) and action spaces (diffusion head for action chunking), similar to the capabilities of the Octo model. 

Any guidance on this matter would be greatly appreciated. Thanks!

Kashu7100, 2024-07-09:
I've also experienced fluctuations in action accuracy when fine-tuning the model.

pxlong, 2024-07-10:
the same issue here #29

kpertsch, 2024-07-12:
Hi everyone, thanks for your questions!
It seems that #29 is closed now -- @pxlong I think #29 was related to some misunderstanding of how our action tokenizer works.

With regards to your original question: we have not experimented with finetuning to new action or observation spaces for OpenVLA, but I would expect it to be possible. Adding proprio should be a small code change to the RLDSTransform. Adding another image input may require a bit more code change. If it's a possibility for you, you could also try "collaging" multiple images into OpenVLAs single-image input simply by concatenating them and resizing the concatenated image to 224x224 -- how reasonable that is depends on your particular setup.

Diffusion head etc is also possible, but not something we have experimented with / have code for.

@marinabar regarding your learning curve it's a bit hard to debug without knowing your data better.
One thing to note is that our loss mask specifically applies loss only on the last 7 action tokens, so if your action space has a different dimension it may not get fully trained?

A simple sanity check could be to train on a very small dataset (eg 10 transitions) and making sure that the model can fully overfit that. If it can't, it's most likely either an issue in the loss masking or your training data is corrupted / impossible to fit.

kpertsch, 2024-07-12:
(feel free to reopen the issue, just closing to keep track what's answered)

closed 2024-07-12 by kpertsch

marinabar, 2024-07-14:
@kpertsch Would you have an example of your accuracy curve during finetuning ? And isn't the mask already based on the taking into account the number of action tokens, as from what I understood it ignores the tokens that are not label predictions, but those labels are already only predicting tokens based on action dimensions and ignoring the rest ? with : ` labels[: -(len(action) + 1)] = self.IGNORE_INDEX` and then 
```
action_gt = batch["labels"][:, 1:].to(action_preds.device)
mask = action_gt > action_tokenizer.action_token_begin_idx
```

I am finetuning it for the Aloha simulation environment, with about 400 frames per episode and 50 episodes.
I have also used the new learning rate you suggested in the updated `finetune.py`, but it did not help.

On 10 trajectories, the policy fully overfitted the data in 200 steps, so the problem does not lie there :/

Would you have some recommendations on what else could be wrong? 
Thanks for the tips !

Zhangwenyao1, 2024-07-25:
> @kpertsch Would you have an example of your accuracy curve during finetuning ? And isn't the mask already taking into account the number of action tokens, as from what I understood it ignores the tokens that are not label predictions, but those labels are already only predicting tokens based on action dimensions and ignoring the rest ? with : ` labels[: -(len(action) + 1)] = self.IGNORE_INDEX` and then
> 
> ```
> action_gt = batch["labels"][:, 1:].to(action_preds.device)
> mask = action_gt > action_tokenizer.action_token_begin_idx
> ```
> 
> I am finetuning it for the Aloha simulation environment, with about 400 frames per episode and 50 episodes. I have also used the new learning rate you suggested in the updated `finetune.py`, but it did not help.
> 
> On 10 trajectories, the policy fully overfitted the data in 200 steps, so the problem does not lie there :/
> 
> Would you have some recommendations on what else could be wrong? Thanks for the tips !

Have you solved this problem?