openvla/openvla #28: Can OpenVLA chat?

psykaunot, 2024-07-09:
First of all, thank you for your great work!
I wanted to ask questions (such as describe what you see) about the model because I saw nothing similar within the paper. Or is the only output we can receive from the model only the 7D action values?

willxxy, 2024-07-09:
From my understanding, in this passage: 

`Unfortunately, the tokenizer used by OpenVLA’s language backbone, the Llama tokenizer [10 ],
only reserves 100 “special tokens” for tokens newly introduced during fine-tuning, which is too
few for the 256 tokens of our action discretization. Instead, we again opt for simplicity and follow
Brohan et al. [7]’s approach by simply overwriting the 256 least used tokens in the Llama tokenizer’s
vocabulary (which corresponds to the last 256 tokens) with our action tokens. Once the actions are
processed into a sequence of tokens, OpenVLA is trained with a standard next-token prediction
objective, evaluating the cross-entropy loss on the predicted action tokens only`

The authors only take out the last 256 "least used" tokens, therefore I believe they append these new action tokens to the existing text tokens. Although the text tokens still exist in the tokenizer, i do not know how much of the catastrophic forgetting phenomenon has taken place when they pretrained the openvla on the autoregressive action token prediction objective. I have tried this though, (e.g., generic robotic setup state input as the image and "Describe the image" as the text) and it outputted 7d action.

siddk, 2024-07-09:
Currently our models cannot chat -- definitely an example of catastrophic forgetting! We didn't co-train on any vision-language data, and if you download the model and just try any text prompt, it'll always try to predict action tokens.

I think co-training, or even training models for less time would be ways to remedy this (and would be super easy to implement in this codebase if desired).

closed 2024-07-09 by siddk

psykaunot, 2024-07-10:
Thank you for both of your answers @willxxy and @siddk ! I believe it will be very useful in the future if we could chat with the model. In any case, I'll keep an eye on the repository. 

Best,