openvla/openvla #29: About finetuning dataset setup

llv22, 2024-07-09:
Hi:

Thanks a lot for you to open-source vla. And I'm trying to use openvla for fine-tuning on my customized dataset.

Before this, I'm doing a verification to check if it can work on parts of droid dataset. Here are what I did:

1. I download rlds_dataset_mod to local
2. I modified prepare_open_x.sh in rlds_dataset_mod to let it download the first top 100 episodes + 4963 trajectories
[image]
3. Then I use the following scripts to start fine-tuning
```bash
export HF_TOKEN="/home/xiandao_airs/.cache/huggingface/token"
torchrun --standalone --nnodes 1 --nproc-per-node 2 vla-scripts/finetune.py \
  --data_root_dir ./data/datasets \
  --dataset_name droid \
  --batch_size 16 \
  --run_root_dir ./run/20240705_12_49 
```

Then I found the loss, accuracy fluctuated quite a lot as shown in the following picture

[image]

So I further debug a little bit

[image]

Then I found the action_decoding result is weird

self.action_tokenizer(action) = '么近巴計客ව食'

Given code implementation in https://github.com/openvla/openvla/blob/main/prismatic/vla/datasets/datasets.py

```python
        prompt_builder = self.prompt_builder_fn("openvla")
        conversation = [
            {"from": "human", "value": f"What action should the robot take to {lang}?"},
            {"from": "gpt", "value": self.action_tokenizer(action)},
        ]
        for turn in conversation:
            prompt_builder.add_turn(turn["from"], turn["value"])

        # Tokenize (w/ `base_tokenizer`)
        input_ids = self.base_tokenizer(prompt_builder.get_prompt(), add_special_tokens=True).input_ids
        labels = list(input_ids)
```

I further checked the loss. And we rely on labels generated by self.action_tokenizer(action) to optimize loss, that's why the loss fluctuating quite a lot.

My question is how to make self.action_tokenizer to convert vectors to natural language to avoid such problem. Correcting me if I'm wrong.

Best Regards
Orlando

Best Regards
Orlando

llv22, 2024-07-11:
@pxlong, I think this issue is because they didn't implement grad_accumulation_steps so the loss fluctuates for each mini-batch. please check their latest submit https://github.com/TRI-ML/prismatic-vlms/commit/7613048828a13e0e4bd95ef22cd19c0c3579d4b4.

llv22, 2024-07-11:
I think I understand the point

closed 2024-07-11 by llv22

Zhangwenyao1, 2024-07-25:
> 我想我明白了

Thanks for your comments！
I meet same problem, can you give me the suggestion and how to solve it.