openvla/openvla #30: Train_loss to become NaN

HanwenWangBUPT, 2024-07-10:
Thank you very much for your outstanding work. I am conducting fine-tuning experiments using the cmu_stretch dataset. My data processing should be fine. I would like to ask if using float16 for computation could cause the train_loss to become NaN. Your code uses bfloat16, but my device, a Titan GPU, does not support this precision.

siddk, 2024-07-10:
Yes - we use `bfloat16` instead of `float16` because [it captures a wider range of values ](https://cloud.google.com/tpu/docs/bfloat16) (basically can represent same range of values as `float32` albeit with less precision).

Unfortunately `float16` captures a smaller range of numbers, so when you train, you are more likely to run into overflow/underflow that leads to the NaN loss. We don't support float16 mixed precision training at this time.

closed 2024-07-10 by siddk

HanwenWangBUPT, 2024-07-11:
Thank you for your response. @siddk