openvla/openvla #31: client cant connect to server?

BankBro, 2024-07-10:
It seems that my server is deployed:
[image]
but it occured something wrong when I running client:
[image]
could you please check it for me?

BankBro, 2024-07-10:
my fault, the first pic is vague...
[image]

Ye13ow77z, 2024-07-10:
the same problem.How to connect to the server?Or to say,I cant give  the request to the server...Did I wrongly operated?The server looked like deployed successfully,but I cant give the client request to the server..Thanks for your addressment!!!Appreciate!!

kpertsch, 2024-07-12:
Hi! Was there any error on the server side when you sent your request?
Please also make sure to run `json_numpy.patch()` on the client side so you can json encode numpy arrays -- it seems that you are currenty converting the numpy array to a list, which isn't ideal. Probably you did this to prevent the error that the missing of `json_numpy.patch()` triggered.

https://github.com/openvla/openvla/blob/7f9467a4880162276f5ef95c047845c08b33960d/vla-scripts/deploy.py#L18

BankBro, 2024-07-13:
@kpertsch Thanks a lot! I found my problem that I use the wrong client IP when connecting to server, which should be 127.0.0.1 instead of 0.0.0.0, because I run the server and client in the same machine. And the program runs well later, appreciating it!

**BTW**, I'd like to know the parameter about `unnorm_key` when I using `requests.post()` at the client. I did not set `unnorm_key`, which should be `None` I think, by browing the code. My code is as follow:
[image]

But the server side telled me I should select one of the followings as `unnorm_key` value:

`AssertionError: Your model was trained on more than one dataset, please pass a unnorm_key from the following options to choose the statistics used for un-normalizing actions: dict_keys(['austin_buds_dataset_converted_externally_to_rlds', 'austin_sailor_dataset_converted_externally_to_rlds', 'austin_sirius_dataset_converted_externally_to_rlds', 'bc_z', 'berkeley_autolab_ur5', 'berkeley_cable_routing', 'berkeley_fanuc_manipulation', 'bridge_orig', 'cmu_stretch', 'dlr_edan_shared_control_converted_externally_to_rlds', 'dobbe', 'fmb_dataset', 'fractal20220817_data', 'furniture_bench_dataset_converted_externally_to_rlds', 'iamlab_cmu_pickup_insert_converted_externally_to_rlds', 'jaco_play', 'kuka', 'nyu_franka_play_dataset_converted_externally_to_rlds', 'roboturk', 'stanford_hydra_dataset_converted_externally_to_rlds', 'taco_play', 'toto', 'ucsd_kitchen_dataset_converted_externally_to_rlds', 'utaustin_mutex', 'viola'])`

And I just use the franka arm for simulation, and I dont konw which scene should it belong to and how to choose a `unnorm_key`? I'm appreciate it if you could help me solve my problem.

kpertsch, 2024-07-13:
Glad you figured it out!
Re unnorm_key: it's unlikely the model will work out of the box, so ideally you'd fine-tune it on your custom dataset (see some explanations here: https://x.com/KarlPertsch/status/1810698405192323088 , instructions for finetuning in the README also) and then set the unnorm_key to the name of your custom dataset.

If you really really want to use the model out of the box you should set the unnorm_key to the dataset that's closest to your test time camera & control stack setup, but again, it would be unlikely to work out of the box.

closed 2024-07-13 by kpertsch

BankBro, 2024-07-13:
@kpertsch Thanks a lot！So nice job!