openvla/openvla #33: Validation

JianJianHeng, 2024-07-11:
Hello, Author,

I have a question that I would like to consult with you. We tried fine-tuning on OpenVLA using our own dataset. We added validation following the training method, but we found that the performance on the training set is excellent, with an action accuracy of 0.85. However, the performance on the validation set is very poor, only around 0.1. I would like to ask if you have encountered a similar situation during your training. Do you have any plans to add validation in the future?

Thank you again for your work.

kpertsch, 2024-07-12:
Hi! Thanks for your question!
Please note that validation accuracy (unfortunately) is usually a very poor metric for policy rollout success. In fact, we often find that policies with high training accuracy and potentially low validation accuracy work best when rolled out on the robot. 
This is also the reason we are not logging validation stats in our train loop, since we didn't find them to be informative in the past.

So I would encourage you to still give your policy a try in rollouts and see whether it works.

As an aside: 85% action token accuracy still seems a bit low -- with small datasets I would expect 95%+ accuracy after finetuning, so you may want to check whether you can further improve accuracy by training longer or cleaning your training data.

closed 2024-07-12 by kpertsch

JianJianHeng, 2024-07-15:
> Hi! Thanks for your question! Please note that validation accuracy (unfortunately) is usually a very poor metric for policy rollout success. In fact, we often find that policies with high training [...]

Thank you for your reply. We will continue training to achieve higher accuracy and then give it a try.

object814, 2024-08-02:
> Hello, Author,
>
> I have a question that I would like to consult with you. We tried fine-tuning on OpenVLA using our own dataset. We added validation following the training method, but we found that [...]

Hi there,
we splited our custom dataset and converted to RLDS as well, however the logging is not showing the validation result, as the author pointed out. I am wondering what changes you made to visualize the validation performance? We wish to see if the same thing happens in our case. Thank you

JianJianHeng, 2024-08-07:
> Hi there, we splited our custom dataset and converted to RLDS as well, however the logging is not showing the validation result, as the author pointed out. I am wondering what changes you made to [...]

Hello,

I apologize for the delayed response due to some personal matters I've been attending to recently. In the original OpenVLA code, there is no part for validation. We created a new validation dataset based on the setting in the dataset that marks train as false. According to the TFDS configuration, it allocates 95% of the data as the training set and 5% as the validation set. We manually coded to obtain this new validation dataset and calculated values such as loss and accuracy on the validation dataset, mimicking the metrics from the training process. However, we later found that the validation metrics were indeed very low, but this did not affect the effectiveness in real-world experiments.

jasonxyliu, 2025-05-12:
Hello @JianJianHeng and @kpertsch, 
Could you please share how you got the validation split when finetuning with a custom dataset? We are trying a different approach, and we would like to examine the validation loss curve.

I set the argument train=False when initiating RLDSDataset, but the returned dataset has the same length as the training dataset. I thought it would be 5% of the whole dataset if we do not do a train-valid split when converting the custom dataset to RLDS format. The function `make_interleaved_dataset` ([Line 525](https://github.com/openvla/openvla/blob/c8f03f48af692657d3060c19588038c7220e9af9/prismatic/vla/datasets/rlds/dataset.py#L525)) seems to return the length of the entire dataset regardless of the train or validation split.

Boltzmachine, 2025-05-28:
> Hello [@JianJianHeng](https://github.com/JianJianHeng) and [@kpertsch](https://github.com/kpertsch), Could you please share how you got the validation split when finetuning with a custom dataset? We [...]

Same issue here. Any solution?