openvla/openvla #34: ValueError: unrecognized configuration class

wuyukun-tong, 2024-07-12:
Traceback (most recent call last):
  File "/home/h666/下载/get_data/examples/depoly.py", line 19, in <module>
    processor = AutoProcessor.from_pretrained("/home/h666/下载/openvla-7b", trust_remote_code=True)
  File "/home/h666/anaconda3/envs/openvla/lib/python3.9/site-packages/transformers/models/auto/processing_auto.py", line 310, in from_pretrained
    return processor_class.from_pretrained(
  File "/home/h666/anaconda3/envs/openvla/lib/python3.9/site-packages/transformers/processing_utils.py", line 465, in from_pretrained
    args = cls._get_arguments_from_pretrained(pretrained_model_name_or_path, **kwargs)
  File "/home/h666/anaconda3/envs/openvla/lib/python3.9/site-packages/transformers/processing_utils.py", line 511, in _get_arguments_from_pretrained
    args.append(attribute_class.from_pretrained(pretrained_model_name_or_path, **kwargs))
  File "/home/h666/anaconda3/envs/openvla/lib/python3.9/site-packages/transformers/models/auto/tokenization_auto.py", line 890, in from_pretrained
    raise ValueError(
ValueError: Unrecognized configuration class <class 'transformers_modules.openvla-7b.configuration_prismatic.OpenVLAConfig'> to build an AutoTokenizer.
Model type should be one of AlbertConfig, AlignConfig, BarkConfig, Bart

kpertsch, 2024-07-12:
Can you give more detail on what you are trying to do here? What is in `/home/h666/下载/openvla-7b`?

wuyukun-tong, 2024-07-12:
thanks for your replay！I sloved that problem.

---- 回复的原邮件 ----
| 发件人 | Karl ***@***.***> |
| 日期 | 2024年07月12日 21:09 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [openvla/openvla] ValueError: unrecognized configuration class (Issue #34) |

Can you give more detail on what you are trying to do here? What is in /home/h666/下载/openvla-7b?

wuyukun-tong, 2024-07-12:
action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)
If I want to deploy the model on a real robot to test its zero-shot ability, how should I set the unnorm_key? Thanks!

kpertsch, 2024-07-12:
Deploying zero-shot will be quite challenging since you'd need to match one of the training dataset's action space, camera setup, controllers etc quite precisely. You'd then set the unnorm_key to whichever dataset is closest to the setup you will evaluate on.
Testing with finetuning is still the easier option for now I believe.

closed 2024-07-12 by kpertsch

wuyukun-tong, 2024-07-12:
I see, thank your very much!

---- Replied Message ----
| From | Karl ***@***.***> |
| Date | 07/12/2024 22:24 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [openvla/openvla] ValueError: unrecognized configuration class (Issue #34) |

Deploying zero-shot will be quite challenging since you'd need to match one of the training dataset's action space, camera setup, controllers etc quite precisely. You'd then set the unnorm_key to whichever dataset is closest to the setup you will evaluate on.
Testing with finetuning is still the easier option for now I believe.

shouji000, 2025-02-21:
excuse me, I encounter with the same problem , can you teach me how to solve the problem "ValueError: unrecognized configuration class "