openvla/openvla #37: RuntimeError: CUDA error: device kernel image is invalid

mellody11, 2024-07-19:
Hi, thanks for your great work! I feel so sorry to disturb you.

When I launch the code `finetune.py`, an error happends at this [line](https://github.com/openvla/openvla/blob/317dbd9e41edaa83ddf2b33f5412b5e7ddadcd4e/vla-scripts/finetune.py#L150): 

```
RuntimeError: CUDA error: device kernel image is invalid
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
```

This error may be caused by a mismatch between the CUDA version and Torch version. However here is my environment:
```
torch                     2.2.0               
torchaudio            2.2.0+cu118            
torchtriton            2.2.0                    
torchvision           0.17.0                 
```
and I install a cuda-nvcc in my virtual environment, so the `nvcc -V` shows:

```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0
```

and finally I run the code `torch.cuda.is_available()` and get `True`.
[image]

I have no idea what to do, looking forward to your reply, sorry again for disturbing, thank you!

Here is the information of my device:
Ubuntu 20.04.4 LTS
GPU: A100
Version:
[image]

kpertsch, 2024-07-22:
Thanks for reaching out!
This seems like a CUDA / torch installation issue that's probably specific to your system and not related to OpenVLA in particular (since others seem to be able to finetune without problems with our script).
One thing I would suggest is to 
(a) verify that this is indeed not an OpenVLA issue by opening a python shell and trying to put any other torch tensor onto GPU -- this should still cause the same issue
(b) wiping your conda environment and reinstalling a fresh conda env with our OpenVLA dependencies

Feel free to reopen if this turns out to be specific to our OpenVLA code!

closed 2024-07-22 by kpertsch

liuxuexun, 2025-04-21:
I update my CUDA Version to 12.4 and solve this problem.

> I have the same problem, and i can successfully run llama3-8b-instruct on cuda:0, but still got the problem as below
> 
> ```
> RuntimeError: CUDA error: device kernel image is invalid
> Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
> ```