openvla/openvla #40: Is there any mini simulation platform to try?

yuedajiong, 2024-07-22:


RZFan525, 2024-07-29:
Do you find a suitable simulation platform?

wangning1234, 2024-08-01:
you can try it with this repo：https://github.com/hilookas/SimplerEnv

betray12138, 2024-08-28:
I'm also eager to know if there are any other simulators available. The training set I'm using is the bridgeV2. Any comments or advice would be appreciated!

moojink, 2024-08-28:
Hi @yuedajiong, @RZFan525, @wangning1234, and @betray12138,

Thank you for your question. We will soon be releasing OpenVLA fine-tuning results for the [LIBERO simulation benchmark](https://libero-project.github.io/main.html), along with instructions on how to reproduce the results. Stay tuned!

-Moo Jin

moojink, 2024-09-05:
Hi all,

We recently added new experiments for fine-tuning OpenVLA on the [LIBERO simulation benchmark](https://libero-project.github.io/main.html) (see Appendix E in [v2 of the OpenVLA paper on arXiv](https://arxiv.org/abs/2406.09246)). We also added instructions to run OpenVLA in simulation and reproduce our results to the README (see the [LIBERO Simulation Benchmark Evaluations](https://github.com/openvla/openvla?tab=readme-ov-file#libero-simulation-benchmark-evaluations) section).

Hopefully this provides a convenient way for you to try OpenVLA in simulation! Please let me know if you run into any issues.

-Moo Jin

closed 2024-09-12 by kpertsch