openvla/openvla #42: I want to create a new dataset, but I don't know what the format of the dataset is. Can you provide an example?

llll111111, 2024-07-24:


maytusp, 2024-07-25:
Hello,

To use the provided finetuning code, your custom data have to be in RLDS format with TFDS interface. Below is the example of how to convert numpy array data to TFDS.

`from typing import Iterator, Tuple, Any
import glob
import numpy as np
import tensorflow as tf
import tensorflow_datasets as tfds
import cv2

class Builder(tfds.core.GeneratorBasedBuilder):
    """DatasetBuilder for example dataset."""

    VERSION = tfds.core.Version('1.0.0')
    RELEASE_NOTES = {
      '1.0.0': 'Initial release.',
    }

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)

    def _info(self) -> tfds.core.DatasetInfo:
        """Dataset metadata (homepage, citation,...)."""
        return self.dataset_info_from_configs(
            features=tfds.features.FeaturesDict({
                'steps': tfds.features.Dataset({
                    'observation': tfds.features.FeaturesDict({
                        'image': tfds.features.Image(
                            shape=(256, 256, 3),
                            dtype=np.uint8,
                            encoding_format='png',
                            doc='camera image.',
                        ),
                        'state': tfds.features.Tensor(
                            shape=(4,),
                            dtype=np.float32,
                            doc='absolute x y z yaw (world coordinate)',
                        )
                    }),
                    'action': tfds.features.Tensor(
                        shape=(4,),
                        dtype=np.float32,
                        doc='dx dy dz dyaw',
                    ),
                    'is_terminal': tfds.features.Scalar(
                        dtype=np.bool_,
                        doc='True on last step of the episode if it is a terminal step, True for demos.'
                    ),
                    'goal': tfds.features.Tensor(
                        shape=(2,),
                        dtype=np.float32,
                        doc='point goal'
                    ),
                }),
                'episode_metadata': tfds.features.FeaturesDict({
                    'file_path': tfds.features.Text(
                        doc='Path to the original data file.'
                    ),
                }),
            }))

    def _split_generators(self, dl_manager: tfds.download.DownloadManager):
        """Define data splits."""
        return {
            'train': self._generate_examples(path='/you_data_dir/train/*.npy'),
            'val': self._generate_examples(path='/you_data_dir//val/*.npy'),
        }

    def _generate_examples(self, path) -> Iterator[Tuple[str, Any]]:
        """Generator of examples for each split."""

        def _parse_example(episode_path):
            # print("episode_path", episode_path)
            # episode_path = tf.io.gfile.GFile(episode_path, mode='r')
            # load raw data --> this should change for your dataset
            
            data = np.load(episode_path, allow_pickle=True)     # this is a list of dicts in our case
            # assemble episode --> here we're assuming demos so we set reward to 1 at the end
            episode = []
            for i, step in enumerate(data):
                action = step['action'].astype(np.float32)
                state = step['state'].astype(np.float32)
                goal = step['goal'].astype(np.float32)
                is_terminal = step['is_terminal']
                cropped_frame = resize_and_crop(step['image'])
                episode.append({
                    'observation': {
                        'image': cropped_frame,
                        'state': state,
                    },
                    'action': action,
                    'is_terminal': is_terminal,
                    'goal': goal,
                })
                action_prev = action
            # create output data sample
            sample = {
                'steps': episode,
                'episode_metadata': {
                    'file_path': episode_path
                }
            }
            # if you want to skip an example for whatever reason, simply return None
            return episode_path, sample

        # create list of all examples
        episode_paths = glob.glob(path)
        # print("episode_paths", episode_paths)

        # for smallish datasets, use single-thread parsing
        for sample in episode_paths:
            # print("sample", sample)
            yield _parse_example(sample)`

kpertsch, 2024-07-26:
Thanks for your answer @maytusp 
For folks who want to convert their own datasets into RLDS, the data format used by our data loader, please check out this repo that walks you through the conversion: https://github.com/kpertsch/rlds_dataset_builder

Alternatively, our finetuning script should work with any torch.data.Dataset that implements the same interface, so you could also skip the RLDS step and try to just implement your own Dataset.

closed 2024-07-26 by kpertsch