openvla/openvla #43: How does it calculate the trajectory between its initial state to its goal position

jis3613, 2024-07-24:
Hi, thanks for open-sourcing the code and providing detailed README to run the code.

I am trying ```the Fine-Tuning OpenVLA via LoRA ``` in README and got into some questions.

It seems that the openvla model is given the image and the language instruction as input
and trained to directly output the final action dimension, which means the final position of the robot (7 action dims).

1. How does the model directly predict the 7 action dims only with the language instruction and the image of the initial environment (state)?
    Doesn't it require any additional images during the robot's execution to go to that final position from its initial position?
    Is there no additional input images during its execution of the action?
    What if something happens and intervenes during the robot's execution? 
    How do you feed back the change of environment ?

3. Given "only" the final action dims by the model such as  [-0.04435408  0.06181145 -0.2151171   0.1507535   0.19060254  0.03201425 0.], 
    how do you execute the robot to go to that position?
    How do you produce the trajectory of the robot to get to that final position from its initial state given only the final goal action?

Thanks

RZFan525, 2024-07-25:
Is the predicted action the final action dimension?
I think it's the next action dimension, which means a trajectory consists of a series action.

jis3613, 2024-07-25:
Well, when I run the fine-tuning code, the model is given input_ids (#batch, #tokens) and pixel_values (#batch, c, h, w) .
As far as I understood, #tokens corresponds to the concatenation of tokenized language instruction and tokenized action_gt.
Then the model outputs corresponding action logits (#bath, #tokens, 32064). 
(32064 is the vocab size of the tokenizer and the model uses the last 265 tokens for the action.)
It argmaxes the logits in the 2nd dim, hence the shape of predicted action is (#batch, #tokens).
When calculating the loss, it masks out the language instruction tokens of the #tokens dimension and calculate loss only on the action tokens.

There is one-to-one mapping between the given input to the output as their batch size is the same,
so I think the model outputs a single action prediction for single input, thus the final action (more specifically the deltas of each 7 action dim to get to the goal position).

If the model outputs a trajectory of a series action, then the model output should be something like (#batch, #trajectory, #tokens).

Now it seems that it only outputs a single action for a single input pair (language instruction, vision).

If there is anything I misunderstood about the code, please correct me and guide.

Thank you.

RZFan525, 2024-07-25:
I'm new in robotics, so I'm not sure if my understanding is true.

I think a trajectory consists of a series of actions, which are called frames.

So when I input the instruction and the start image to the OpenVLA model, it outputs an action, which can make the robot move to the next frame. Then, I can input the instruction and the current image to the model and the model will output the next action. We can repeat this process util the state reaches the goal.

If there is anything I misunderstood about the code, please correct me and guide.

jis3613, 2024-07-25:
Well that is what I thought in the first place, but after reading and running the code, it seems not.
If the model is predicting the very next action given the image and the language instruction of the current time step, 
How far/much the action output of the model moves the robot given only a single frame of image?

RZFan525, 2024-07-25:
I see. But, if the output action is the goal action, how do we apply it to the robot if it is so far between the start and goal states?

jis3613, 2024-07-25:
That is what I am curious about. Maybe I am misunderstanding something here. 
I hope the authors can answer this question.
Maybe apply inverse-kinematics method to enable motion planning of the robot to the goal positon? Idk haha

RZFan525, 2024-07-25:
Yes me too, Hoping the authors can answer

kpertsch, 2024-07-26:
Hi folks,
Thanks for your interest in OpenVLA and excited that you are getting started in robotics! :)
@RZFan525 your understanding is correct: the model indeed outputs a single action given the *current* image and the language instruction. This 7-dim action is a single timestep's delta in end-effector pose, so it's a *small* step (typically what we want the robot to execute in 0.2 seconds or so). 

Now to make the robot move and solve a task, we will rerun this policy prediction 5-10 times a second, always feeding in the current image from the camera and producing the next small action, which we send to the robot controller to execute. If we do this over and over, the policy will eventually solve the task. But, to be clear, the policy is not only called once per task execution, it is called iteratively.

closed 2024-07-26 by kpertsch

jis3613, 2024-07-26:
Thanks for the clarification!
I have some following questions.
Now I understand that the current images are iteratively fed into the policy to predict the 7-dim action deltas.
Then what about the language instruction ? Is it only fed into the policy once at the begining of the task execution?

Also, the 7-dim action is a single timestep's delta in end-effector pose, how do you control the movements of other joints of the robot?
Does the robot controller automatically calculate the translation and rotation of the other joints?

Thanks!

kpertsch, 2024-07-26:
The language instruction is passed into the policy in every step, but we typically keep it constant throughout a task trajectory.

For controlling the robot, we are using IK control (inverse kinematics) -- maybe you can google what that is. But basically it means we have an underlying controller that knows about the kinematic chain of the robot (ie the joints and their connections) and can analytically map delta end-effector commands into delta-joint angle commands (which in turn get mapped into actual motor currents by an underlying tracking controller).

This kind of setup is very standard in robot learning and not anything specific to the OpenVLA policy. It operates one layer of abstraction higher and just assumes that the robot can be commanded via delta end-effector commands.

jis3613, 2024-07-26:
Thank yo so much!
Now things are more clear for me now!

RZFan525, 2024-07-26:
Thank you for your detailed explanation, which makes me more clear.

I have an additional question: this iterative process is too low-efficiency to be applied in a realistic scene, so what are the advantages compared with other algorithms?

kpertsch, 2024-07-26:
One benefit of such a "closed loop" policy is that it can react to any mistakes it makes along the way: even if one step is not ideal, the next step can look at the resulting state and correct.
A policy that predicts all actions for the whole trajectory in the beginning ("open loop") cannot correct and tends to work worse.

In practice, many modern imitation learning methods do something in between, where they predict chunks of actions at once, with 8-32 actions. Testing such action chunking for OpenVLA would be a good thing to try.

RZFan525, 2024-07-26:
Thank you for your reply, which enables me to understand VLA models more deeply.

TousakaNagio, 2024-10-06:
> Hi folks, Thanks for your interest in OpenVLA and excited that you are getting started in robotics! :) @RZFan525 your understanding is correct: the model indeed outputs a single action given the [...]

Thank you for your explanation, but I am wondering why is the the pixel_values from bridgedata v2 is in the shape of (6, 256, 256)? Is the LoRA finetuning code conditioned on the goal image and current input frame?

esychoi, 2024-10-08:
Would it make sense to put the primary image as the first 3 dimensions and the secondary image as the last 3 dimensions if I want to use several camera angles ?

TousakaNagio, 2024-10-08:
After my investigation, they are the same images with different image tranform applied for SigLip and DinoV2. According to @jis3613 , they applied two vision encoders for visual representation. They feed pixel_values in shape of (6, 256, 256) to vision tower, that is why I was wondering why there are 6 channels at fisrt. The shape of pixel_values is (3, 256, 256) before feeded into image transformation. That is why I conclude that the applied different image tranformation for each vision encoder.