openvla/openvla #45: About how to identify the instruction is done.

BankBro, 2024-07-24:
I have acknowledged that the predicted action is in the form of delta. And it is necessary to predict a certain number of times to finish the instruction. So each time I get a new predicted action. But how do I know if the task is complete? When it is time for the openvla to stop, what would the model return, and how to identify the task has been done? Thanks!

RZFan525, 2024-07-25:
I have the same question.

Subuday, 2024-07-25:
At paper presentation he said that model has not been trained to produce done output.
So you can implement somewhat aka timer, or fine-tune the model on finished videos where the robot is finishing in stale position

BankBro, 2024-07-25:
> At paper presentation he said that model has not been trained to produce done output. So you can implement somewhat aka timer, or fine-tune the model on finished videos where the robot is finishing in stale position

Thanks! I miss it!

kpertsch, 2024-07-26:
Jup, @Subuday's answer is correct -- the current model is not trained to predict when it's done solving a task. 
Your options are:
1. implement a manual timeout (ie terminate the task after a fixed number of steps or via manual human intervention)
2. fine-tune the model on a dataset that shows the robot not moving after the task is done -- then the policy will naturally stop after completing a task
3. introduce an additional dimension to the action space during fine-tuning that corresponds to a "done" action and is only set to 1 at the end of the trajectory and 0 otherwise -- then you can use the value of this action at inference time to determine when the policy thinks that the task is done.

In our experiments we used a combination of 1 and 2.

closed 2024-07-26 by kpertsch