openvla/openvla #46: About memory growth

Toradus, 2024-07-24:
There are several issues where people ask why the memory keeps growing when using the Dataloader:
https://github.com/openvla/openvla/issues/4
https://github.com/octo-models/octo/issues/16

i kind of want to reopen the question with one assumption that maybe someone can verify:
The restructuring of trajectories is done with tf's symbolic tensors. Since we randomly access samples of a trajectory (random sharding of tfds before accessing it) the data of a trajectory is not loaded sequentially. When loading data with history or future_window_size, the previous/next samples are also loaded due to the trajectory transform executing while accessing the sample.

Does TF cache these previous / next samples and reuses them once the corresponding sample is loaded?
Could that be why the memory is growing?

I've noticed that the memory keeps growing till a specific point, then goes slightly up and down, which is kind of annoying when loading many different cameras.
If that is the case, is there a way to disable this caching?

Maybe this is also the wrong repository for this, i will prbl cross post it to Octo but i think they are currently on a conference

kpertsch, 2024-07-26:
Hi! 
TFDS memory management is annoyingly pretty much a black box. I think the hypothesis you're proposing does not apply in our case since we train the model without history and without future action steps (ie we load a single timestep with image and one action).

@kvablack has spent a lot of time when developing the Octo data loader to try and get memory consumption down. For OpenVLA, things get a bit worse since PyTorch forces us to make one data loader per device, so memory usage gets N-times larger for N devices. We haven't really found a way around this with TFDS (and this wasn't a blocker for our experiments).

If this is holding your experiments back, our codebase should be compatible with any custom torch.data.Dataset that you create for your data -- we just stuck with TFDS since that's what the OpenX datasets are saved in that we used for this project. But if you're doing some training on custom data, making a custom properly-PyTorch data loader may not be a bad option.

closed 2024-07-26 by kpertsch

Toradus, 2024-07-26:
Thanks for the response!
Thought i had finally tracked down the problem, but as you stated, when training without history / future the memory still shows this behavior it can't be the problem...

I currently want to utilize the OpenX dataset, therefore switching to a PyTorch dataloader is not really possible (except if i download all data).
I did some investigations regarding speed / memory consumption tradeoffs and for me, using num_workers = 1 (in the PyTorch part) with prefetching and dispatching batches outperformed instantiating one dataloader per devide (single node, 4 GPUs) in both aspects. The PyTorch documentation also states that it is possible to modify the workers creation to not copy the dataset, so maybe this can be further modified to decrease the parallelism in TFDS (which hopefully has positive impact on the memory management), but increase the parallism in PyTorch.
As i didn't want to go further down the rabbit hole, i didn't try this.

Thanks again for the greate work on the dataloader, i think with tools like this, we have a bright future ahead!

LukeLIN-web, 2025-03-21:
> Thanks for the response! Thought i had finally tracked down the problem, but as you stated, when training without history / future the memory still shows this behavior it can't be the problem...
>
> I [...]

Thank you for your insight. Which code you modified for `num_workers = 1 `? How to prefetching and dispatching batches?