openvla/openvla #48: Finetune for bimanual robot

jellyho, 2024-07-26:
Thanks for open sourcing your great work!

I want to fine tune openvla using bimanual dataset, and generated dataset for it.
But I guess in the /prismatic/vla/datasets/rlds/oxe/materialize.py said only EEE_POS or EEE_R6 action is supported.

Is it possible to finetune with dataset that action dimension is 14? Or which part I have to modify?

Best
Hokyun

WT-MM, 2024-07-26:
Had the same question about the allowed action encoding scheme. I notice that some of the oxe datasets use JOINT_POS and JOINT_POS_BIMANUAL, but the config generator in materialize.py doesn't seem to allow that.

kpertsch, 2024-07-26:
We only pre-train OpenVLA with single-arm end-effector control data -- so finetuning with a different action space is a bit "at your own risk" :D
But generally, I would expect this to work reasonably. You would just need to define a new action encoding with your 14-dim actions and make sure that when you finetune you apply loss to all 14 action tokens (and sample 14 tokens at inference time ofc)

closed 2024-07-29 by kpertsch