openvla/openvla #54: 2 different ways of loading model.

willxxy, 2024-08-01:
Hello,

Thank you for the great works again. I had a question on the way we are supposed to load the model after finetuning.

After finetuning, we have the `adapter_weights` folder with the `adapter_config.json` and `adapter_model.safetensors`.
Additionally, we have the `openvla-7b+kinova_dataset+b16+lr-0.0005+lora-r32+dropout-0.0` folder with the openvla configs and safetensors.

I saw that there is this way of loading the model for inference

```
adapter_weights_path = './adapter_weights/'
open_vla_weights_path = 'openvla-7b+kinova_dataset+b16+lr-0.0005+lora-r32+dropout-0.0'
base_vla = AutoModelForVision2Seq.from_pretrained(open_vla_weights_path, trust_remote_code=True, torch_dtype=torch.bfloat16, low_cpu_mem_usage=True)
merged_vla = PeftModel.from_pretrained(base_vla, adapter_weights_path)
```

And this way, which is shown throughout the repo and in this [tutorial](https://x.com/KarlPertsch/status/1810698405192323088).

```
open_vla_weights_path = 'openvla-7b+kinova_dataset+b16+lr-0.0005+lora-r32+dropout-0.0'
base_vla = AutoModelForVision2Seq.from_pretrained(open_vla_weights_path, trust_remote_code=True, torch_dtype=torch.bfloat16, low_cpu_mem_usage=True)
```

The latter method does not include the lora weights so wouldn't the former method the correct way to inference the model?
Would you be able to clarify this a bit?

Thank you.

siddk-tri, 2024-08-01:
Sorry for the confusion!

At the [end of the `finetune.py` script](https://github.com/openvla/openvla/blob/main/vla-scripts/finetune.py#L310-L319), we explicitly "merge" the LoRA adapter weights into the VLA weights, and save a "unified" checkpoint; this allows you to just load the adapted model via `.from_pretrained()` as in the second example above.

The first example is only necessary when the adapter weights have *not* been merged into the base VLA!

siddk, 2024-08-01:
Sorry for the confusion!

At the [end of the `finetune.py` script](https://github.com/openvla/openvla/blob/main/vla-scripts/finetune.py#L310-L319), we explicitly "merge" the LoRA adapter weights into the VLA weights, and save a "unified" checkpoint; this allows you to just load the adapted model via `.from_pretrained()` as in the second example above.

The first example is only necessary when the adapter weights have *not* been merged into the base VLA!

closed 2024-08-01 by siddk