openvla/openvla #55: Question about augmentation during inference.

willxxy, 2024-08-01:
In [Issue 12](https://github.com/openvla/openvla/issues/12) you mentioned to only apply center crop to mirror the training if trained with augmentation. However, I see that you apply these augmentations as well

```
            rlds_config["frame_transform_kwargs"].update({"image_augment_kwargs" : dict(
                random_resized_crop=dict(scale=[0.9, 0.9], ratio=[1.0, 1.0]),
                random_brightness=[0.2],
                random_contrast=[0.8, 1.2],
                random_saturation=[0.8, 1.2],
                random_hue=[0.05],
                augment_order=[
                    "random_resized_crop",
                    "random_brightness",
                    "random_contrast",
                    "random_saturation",
                    "random_hue",
                ],
            )}),
```
I have tried the center cropping mentioned in the Issue 12 but it did not work well. So I want to know if we should  utilize the full augmentation as well during inference?

Thank you.

willxxy, 2024-08-02:
I have also tried the given augmentations and the performance is still not great.

kpertsch, 2024-08-07:
Hi thanks for your question!
We do indeed apply all these augmentations during fine-tuning by default (please check that you set the `image_aug` argument (https://github.com/openvla/openvla/blob/317dbd9e41edaa83ddf2b33f5412b5e7ddadcd4e/vla-scripts/finetune.py#L86)[here] to `True`). Note that we do not use augmentation during pre-training, but I assume that you are not using the model 0-shot, but are fine-tuning it on some of your own data?

If you have issues getting the model to work, you can try to first train it on a very simple task with a very small dataset to sanity-check your fine-tuning and inference setup. Here is a small thread that gives some best practices: https://x.com/KarlPertsch/status/1810698405192323088

closed 2024-08-07 by kpertsch