openvla/openvla #56: Question about potential mis alignment between finetuning and inference preprocessing

object814, 2024-08-03:
Hi! First of all thank you for your great work.

I tried finetuning OpenVLA on my custom dataset, I successfully done the following:
1. converted my own dataset into RLDS dataset using the [RLDS builder](https://github.com/kpertsch/rlds_dataset_builder) suggested in the tutorial, and registered the dataset in configs.py and transform.py
2. finetuned OpenVLA without image augmentation since I am working in a simulation environment, and reached action_accuracy near 1.0 for training
3. during inference however, I process the raw input like this:
`image_vis = PIL.Image.fromarray(image)`
`prompt = f"In: What action should the robot take to {task_language_instruction}?\nOut:"`
 `with torch.no_grad():`
     `inputs = processor(images=image_vis, text=prompt, return_tensors="pt").to(f"cuda:{DEVICE_ID}", dtype=torch.bfloat16)`
     `action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)`
the finetuned OpenVLA was not able to perform at all on the training task with different rollout. I read acroos some other issues and found out other people are experiencing the same problem, your answer was to make sure all the preprocess aligns exactly the same for training and inference.

However, I am struggling to find out where this misalignment happened. Because I noticed that:
i) the [RLDS builder](https://github.com/kpertsch/rlds_dataset_builder) did some preprocessing on images and text inside
ii) inside OpenVLA finetune code (openvla/prismatic/vla/datasets/datasets.py), there are some processing to convert RLDS dataset format into OpenVLA default format
I assume that, the OpenVLA itself should not have misalignment during training and inference so perhaps ii) is not the problem? Do you think I have to look into i) to find out the preprocessing RLDS did on my data? Or am I missing out something.

Appreciate your time and wish you all the best :)

object814, 2024-08-04:
Also, I disabled the image augmentation during finetuning.

object814, 2024-08-05:
I am now considering it's the problem here when I set unnorm_key unchanged as "bridge_orig":
`action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)`
I traced the unnorm_key and ended up in prismatic/vla/datasets/rlds/oxe/mixtures.py, it's also mentioned in "Fully fine-tune OpenVLA" section in README. However in "Fine-Tuning OpenVLA via LoRA" it is not mentioned.
I am wondering is this the problem or with just fine-tuning via LoRA I can still stick to unnorm_key as "bridge_orig"?

object814, 2024-08-05:
New update, I checked [this issue](https://github.com/openvla/openvla/issues/32) and found that I have to manually add configuration of my custom dataset into openvla config
After I did this, the output seemed more reasonable, but it is still not performing very well, I will keep looking into it :)

kpertsch, 2024-08-07:
Ah sorry just saw this -- yes, you need to manually make sure that you have the right unnormalization constants at the moment when finetuning unfortunately :I
Regarding the image inputs, a good strategy can be to pdb into the model training code right before your batch hits the model and visualize the inputs / save some to disk. Then do the same in your inference code right before you feed the image to the model and make sure the two inputs look comparable (eg no normalization issues etc). hope this helps!

closed 2024-08-07 by kpertsch

object814, 2024-08-07:
> Ah sorry just saw this -- yes, you need to manually make sure that you have the right unnormalization constants at the moment when finetuning unfortunately :I Regarding the image inputs, a good [...]

Thanks!