openvla/openvla #62: Issue with Repeating Model Output After Fine-Tuning

JHKim-snu, 2024-08-12:
Hello,

Thank you for your excellent work. I have a question regarding the fine-tuned model's performance. I fine-tuned the model on a dataset of around 100 episodes as per your guidelines. While the training loss decreased as expected, I encountered an issue during testing. On our robot, the model outputs identical actions consecutively throughout an episode, producing the same action. Have you experienced this behavior before? 

Thanks again for your valuable work.

JHKim-snu, 2024-08-13:
@EmbodiedSolutions I'm sorry for the confusion. I was referring to "fine-tuning" with the provided guidelines, not guidelines for "using a dataset".
 
I have collected my own dataset and converted it into RLDS format following the instructions in [this repo](https://github.com/kpertsch/rlds_dataset_builder).

EmbodiedSolutions, 2024-08-14:
@JHKim-snu No, actually I knew you were referring to fine-tuning; I should have said "using a dataset of 100 episodes for fine-tuning" to be clearer. Thanks for responding!

kpertsch, 2024-08-22:
Hi!
To confirm: did you train your model to predict a single action at a time or an action chunk? The repeating action problem can be quite common for action chunks, but I would be surprised if it's common for single action prediction with various inputs.

If you want to debug this, a simple step would be to load up some images from the training data in your inference pipeline and pretend that they came from the robot camera. If everything is set up correctly the model should output the correct actions from the training data for the training images.

closed 2024-08-22 by kpertsch