openvla/openvla #69: finetune (lora) with bridge_orig not work

AnonymousUser360, 2024-08-19:
When I put the downloaded bridge data following the instruction in ./openvla/BridgeData/bridge_data/1.0.0/*datas*.
Then run finetune.py, it occurs error in this function : vla_dataset = RLDSDataset().

The error is:
 Did you mean: bridge_orig -> bridge ?

No builder could be found in the directory: /mnt/nfs2/xx/code/openvla/BridgeData for the builder: bridge_orig.
No registered data_dirs were found in:
	- /mnt/nfs2/xx/code/openvla/BridgeData
.

And I don't find bridge_orig in "tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset bridge_orig not found.
Available datasets:
	- abstract_reasoning
	- accentdb"
	
Anyone can help?  Thanks a lot

MagdalenaKotynia, 2024-08-21:
@AnonymousUser360 I had the same problem. I have replaced `bridge_orig` with `bridge` everywhere in the code and it started working.

AnonymousUser360, 2024-08-21:
> @AnonymousUser360 I had the same problem. I have replaced `bridge_orig` with `bridge` everywhere in the code and it started working.

Thanks, I also find it, especially should keep the dataset name consistent with the downloaded dataset json file.

lucywang720, 2024-09-01:
hi, I would like to know which folders should I change bridge to bridge_orig, and whether I should also change similar 'bridge_oxe' to 'bridge_orig_oxe'.

Thank you very much!