openvla/openvla #74: The action accuracy after fine-tuning does not reach ~0.95

HanwenWangBUPT, 2024-08-21:
Thank you for your outstanding work. I am currently fine-tuning OpenVLA based on LoRA using a simple grasping action dataset that I collected myself(~100 Trajectories). However, during training, I am unable to achieve the action accuracy (~0.95) that you mentioned. Although my training loss has converged, I would like to understand what potential issues might be causing this.

Fine-tuning data example
[image]

Fine-tuning Parameters
`   batch_size: int = 1                                          
    max_steps: int = 200_000                                        
    save_steps: int = 5000                                          
    learning_rate: float = 1e-5                                    
    grad_accumulation_steps: int = 16                              
    image_aug: bool = True                                        
    shuffle_buffer_size: int = 100_000                              
`
Fine-tuning status
[image]

kpertsch, 2024-08-22:
Hi!
Thanks for trying out OpenVLA! Your robot setup looks great!
I am a bit surprised that the model doesn't fit the data. One thing that's a bit weird is that your accuracy curve has these very discrete values (it looks like it's computed over 4 action samples only?) and seems very noisy. Maybe one thing to try would be to increase the batch size of your run? it looks like you may have a bug in how the batch size of your run is computed?

kpertsch, 2024-08-27:
Will close for now, feel free to reopen!

closed 2024-08-27 by kpertsch

HanwenWangBUPT, 2024-08-29:
Thank you for your response. I did encounter some bugs with the `batch_size` settings, but I've resolved them now.

YEOMJINSEOP, 2024-11-11:
Hi, @HanwenWangBUPT  i'm also experiencing noise in experiment results. Could you share how you handle batch_size setting? Thanks!

HanwenWangBUPT, 2024-11-12:
> Hi, @HanwenWangBUPT i'm also experiencing noise in experiment results. Could you share how you handle batch_size setting? Thanks! 
If your device's GPU memory cannot support a large `batch_size`, you need to ensure that `grad_accumulation_steps` set to 16, also you need to ensure that your code successfully passes the parameter. Hope this can help you.