openvla/openvla #75: Questions about Training Data

MedesL, 2024-08-22:
Hello guys, 

I have few questions about the training data. 
At present, the data of my robot arm include the attitude of the end, XYZRPY, opening and closing Angle. I really confused that how to find the delta or Hertz quantity? Like how do we determine the frequency? 

And thank you very much for authors' amazing innovation.

kpertsch, 2024-08-26:
Hi thanks for reaching out!
I am not sure I fully understand your question -- your robot system usually dictates the control frequency. E.g. what is the frequency at which you collected your data during teleoperation? That would tell you the frequency of your data and then you would compute delta actions either as the actions that your teleop device sent to the robot (if you are using non-blocking control) or as the delta between consecutive states (if you are using blocking control). 
I hope this helps to answer your question!

closed 2024-08-26 by kpertsch

ldj210, 2024-08-27:
Thank you team for a great job. I have a question. The output of the model is the delta quantity of the action, and the input should also be the delta quantity. The definition of my action is x, y, z, r, p, y and the opening and closing angle of the end effector. My question is about the processing of the delta amount of the action. Do I process it myself after collecting the data? But when I was reading the code, I found that prismatic/vla/datasets/rlds/dataset.py in this script has the function make_dataset_from_rlds() and absolute_action_mask. After I printed this, I found that it is [false,false,false,false,false, false, true], isn't this part dealing with the delta amount of the action? A little confused and looking forward to your reply.
[image]
[image]