openvla/openvla #76: Question about clip on action tokenizer

LingwuPro, 2024-08-23:
The de-tokenizer implementation for action in the code is done by forced truncation, that is, if the prediction result is outside the bins, it is forcibly truncated to min_action. I think this is a very rough way of processing, so why not filter by intercepting the softmax of the last 256 words output by llama-8B, and retain 32064 word list outputs, and truncate if it exceeds the boundary?

LingwuPro, 2024-08-23:
```
def decode_token_ids_to_actions(self, action_token_ids: np.ndarray) -> np.ndarray:
        """
        Returns continuous actions for discrete action token IDs.

        NOTE =>> Because of the way the actions are discretized w.r.t. the bins (and not the bin centers), the
                 digitization returns bin indices between [1, # bins], inclusive, when there are actually only
                 (# bins - 1) bin intervals.

                 Therefore, if the digitization returns the last possible index, we map this to the last bin interval.

        EXAMPLE =>> Let's say self._bins has 256 values. Then self._bin_centers has 255 values. Digitization returns
                    indices between [1, 256]. We subtract 1 from all indices so that they are between [0, 255]. There
                    is still one index (i==255) that would cause an out-of-bounds error if used to index into
                    self._bin_centers. Therefore, if i==255, we subtract 1 from it so that it just becomes the index of
                    the last bin center. We implement this simply via clipping between [0, 255 - 1].
        """
        discretized_actions = self.tokenizer.vocab_size - action_token_ids
        discretized_actions = np.clip(discretized_actions - 1, a_min=0, a_max=self.bin_centers.shape[0] - 1)

        return self.bin_centers[discretized_actions]
```
change the `model.generate()` may be better that just use `np.clip()` in action tokenizer.

kpertsch, 2024-08-26:
In practice I don't think the finetuned model will output tokens outside the range of valid action tokens -- since it has been trained exclusively with tokens in that range during action finetuning the model very quickly latches on to the correct range. So I don't think this is super important in practice.

If you want a fancier decoding strategy you can take a look at the RT-2 paper -- I believe there the decoding during beam search was limited to the correct range of action tokens.

closed 2024-08-26 by kpertsch

esychoi, 2024-10-07:
Does that mean that there might be a problem with the finetuning itself if my model does not output tokens in the valid range (between 31,808 and 32,064) ?