openvla/openvla #83: LORA fine tuning inside a container:  The client socket has failed to connect to any network address

akaraliNV, 2024-08-29:
I am trying to finetune with lora inside a container, but the script throw that The client socket has failed to connect to any network address

```
(openvla) root@6968782:/workspace/openvla# torchrun --standalone --nnodes 1 --nproc-per-node 1 vla-scripts/finetune.py   --vla_path "/workspace/openvla/openvla-7b"   --data_root_dir "/workspace/openvla/data"   --dataset_name bridge_orig
/root/miniconda3/envs/openvla/lib/python3.10/site-packages/_distutils_hack/__init__.py:55: UserWarning: Reliance on distutils from stdlib is deprecated. Users must rely on setuptools to provide the distutils module. Avoid importing distutils or import setuptools first, and avoid setting SETUPTOOLS_USE_DISTUTILS=stdlib. Register concerns at https://github.com/pypa/setuptools/issues/new?template=distutils-deprecation.yml
  warnings.warn(
2024-08-29 20:02:34.172269: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-08-29 20:02:34.172370: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-08-29 20:02:34.173312: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-08-29 20:02:34.179339: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-08-29 20:02:34.943015: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
**[W socket.cpp:697] [c10d] The client socket has failed to `connect `to ?UNKNOWN? (errno: 110 - Connection timed out).
[W socket.cpp:697] [c10d] The client socket has failed to connect to ?UNKNOWN? (errno: 110 - Connection timed out).
[E socket.cpp:753] [c10d] The client socket has failed to connect to any network address of (6968782, 41093).**
Traceback (most recent call last):
  File "/workspace/openvla/vla-scripts/finetune.py", line 42, in <module>
    from prismatic.models.backbones.llm.prompting import PurePromptBuilder, VicunaV15ChatPromptBuilder
  File "/workspace/openvla/prismatic/__init__.py", line 1, in <module>
    from .models import available_model_names, available_models, get_model_description, load
  File "/workspace/openvla/prismatic/models/__init__.py", line 1, in <module>
    from .load import available_model_names, available_models, get_model_description, load, load_vla
  File "/workspace/openvla/prismatic/models/load.py", line 16, in <module>
    from prismatic.models.materialize import get_llm_backbone_and_tokenizer, get_vision_backbone_and_transform
  File "/workspace/openvla/prismatic/models/materialize.py", line 12, in <module>
    from prismatic.models.backbones.llm import LLaMa2LLMBackbone, LLMBackbone, MistralLLMBackbone, PhiLLMBackbone
  File "/workspace/openvla/prismatic/models/backbones/llm/__init__.py", line 1, in <module>
    from .base_llm import LLMBackbone
  File "/workspace/openvla/prismatic/models/backbones/llm/base_llm.py", line 33, in <module>
    overwatch = initialize_overwatch(__name__)
  File "/workspace/openvla/prismatic/overwatch/overwatch.py", line 147, in initialize_overwatch
    return DistributedOverwatch(name) if int(os.environ.get("WORLD_SIZE", -1)) != -1 else PureOverwatch(name)
  File "/workspace/openvla/prismatic/overwatch/overwatch.py", line 54, in __init__
    self.logger, self.distributed_state = ContextAdapter(logging.getLogger(name), extra={}), PartialState()
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/accelerate/state.py", line 211, in __init__
    torch.distributed.init_process_group(backend=self.backend, **kwargs)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/c10d_logger.py", line 86, in wrapper
    func_return = func(*args, **kwargs)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/distributed_c10d.py", line 1177, in init_process_group
    store, rank, world_size = next(rendezvous_iterator)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/rendezvous.py", line 246, in _env_rendezvous_handler
    store = _create_c10d_store(master_addr, master_port, rank, world_size, timeout, use_libuv)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/rendezvous.py", line 174, in _create_c10d_store
    return TCPStore(
torch.distributed.DistNetworkError: The client socket has failed to connect to any network address of (6968782, 41093). The client socket has failed to connect to ?UNKNOWN? (errno: 110 - Connection timed out).
[2024-08-29 20:07:02,381] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: 1) local_rank: 0 (pid: 68404) of binary: /root/miniconda3/envs/openvla/bin/python
Traceback (most recent call last):
  File "/root/miniconda3/envs/openvla/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 347, in wrapper
    return f(*args, **kwargs)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 812, in main
    run(args)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 135, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
vla-scripts/finetune.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2024-08-29_20:07:02
  host      : 6968782
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 68404)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
```

kpertsch, 2024-09-02:
This problem doesn't seem directly related to OpenVLA, but a more generic issue with running PyTorch DDP in your environment. 

Could you check whether the training runs for you outside of docker on the same machine if that's an option?
I googled your error and it seems that you may need to open a certain port to be accessible for the train script, so it can use it for synchronization between the different processes in distributed training. If this continues to be a problem, maybe you can raise it in the PyTorch forum or google how to run PyTorch DDP with docker. I assume you'd encounter the same error if you tried to run the PyTorch DDP tutorial example in your Docker (ie it's not related to something OpenVLA specific).

Btw, if you want to only train on single GPU (like your current command indicates) you could run the finetuning script without torchrun, simply by calling it directly via `python vla-scripts/finetune.py ...` that may avoid the headache with process communications.

closed 2024-09-02 by kpertsch

HeinzYan, 2024-12-06:
Hi, have you solved this problem? I met the same problem as you