openvla/openvla #85: Issues about bridge dataset setup

lucywang720, 2024-09-01:
Hi,

I met this error

```
The builder directory /cluster/data7a/whq/data/bridge_orig doesn't contain any versions.
No builder could be found in the directory: /cluster/data7a/whq/data for the builder: bridge_orig.
No registered data_dirs were found in:
        - /cluster/data7a/whq/data
```

i change the 'bridge' in oxe to 'bridge_orig' but this error remains. I wonder if you could help me solve this error.

Thanks!

lucywang720, 2024-09-01:
my running comman:

```
torchrun --standalone --nnodes 1 --nproc-per-node 8 vla-scripts/train.py \
  --pretrained_checkpoint /cluster/nvme9a/whq/models--openvla--openvla-7b-prismatic/snapshots/5e44aaf23f992e150f26b257500144225ab6643b/checkpoints/step-295000-epoch-40-loss=0.2200.pt \
  --vla.type prism-dinosiglip-224px+mx-bridge \
  --data_root_dir /cluster/data7a/whq/data \
  --vla.data_mix bridge_orig \
  --run_root_dir /cluster/sata9/whq/openvla/log \
  --run_id Full \
  --image_aug False \
  --wandb_project openvla \
  --save_interval 10 \
  --is_resume True \
  --hf_token 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 \
  --resume_step 295000 \
  --resume_epoch 40
```

lucywang720, 2024-09-01:
file list:

/cluster/data7a/whq/data$ ls -R
.:
bridge_orig

./bridge_orig:
1.0.0

./bridge_orig/1.0.0:
 action_proprio_stats_4203a9a0c2e65f37c33c477d42e8f9d87929e12a4df4e51dbf1479f417d7f3e5.json   bridge_dataset-train.tfrecord-00576-of-01024
 action_proprio_stats_7d6a416829d818b733e7342f225f3c522a8265a5224e0175f2ab28e26a932ff1.json   bridge_dataset-train.tfrecord-00577-of-01024
 action_proprio_stats_9ccaabde9b86d1b6d3bfdf6fcfc1ab326e3265b928ae4dd4cab47e2067b9ecab.json   bridge_dataset-train.tfrecord-00578-of-01024
 action_proprio_stats_eb41baf7d978d166b787f61ef00e0c6a1b747c3dc68e2d19167bf750b025c417.json   bridge_dataset-train.tfrecord-00579-of-01024

kpertsch, 2024-09-02:
I am not 100% sure where this error is coming from -- could you post a longer stack trace for the error?

As a first debugging step, could you try loading the dataset by simply running:
```
import tensorflow_datasets as tfds
ds = tfds.load("bridge_orig", data_dir="/cluster/data7a/whq/data", split="train")
```
If this throws an error, something about the raw data is broken. Also to make sure: I assume there are more files inside the bridge_orig/1.9.0 directory, right? It seems like there should be 1024 shard files and a file called `dataset_info.json` and one called `features.json`.

P.S.: ideally please don't post your huggingface token on a publicly visible forum like this, it would allow others to access your account.

lucywang720, 2024-09-02:
Thank you very much for your detailed explanation! 
i run this command and get this error, but I'm not sure if this is what you mean by "broken raw data":

```
Python 3.9.19 | packaged by conda-forge | (main, Mar 20 2024, 12:50:21) 
[GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow_datasets as tfds
>>> ds = tfds.load("bridge_orig", data_dir="/cluster/data7a/whq/data", split="train")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/logging/__init__.py", line 166, in __call__
    return function(*args, **kwargs)
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/load.py", line 633, in load
    dbuilder = _fetch_builder(
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/load.py", line 488, in _fetch_builder
    return builder(name, data_dir=data_dir, try_gcs=try_gcs, **builder_kwargs)
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/logging/__init__.py", line 166, in __call__
    return function(*args, **kwargs)
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/load.py", line 215, in builder
    raise not_found_error
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/load.py", line 196, in builder
    cls = builder_cls(str(name))
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/load.py", line 121, in builder_cls
    cls = registered.imported_builder_cls(str(ds_name))
  File "/cluster/home2/whq/miniconda3/envs/openvla/lib/python3.9/site-packages/tensorflow_datasets/core/registered.py", line 301, in imported_builder_cls
    raise DatasetNotFoundError(f'Dataset {name} not found.')
tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset bridge_orig not found.
Available datasets:
        - abstract_reasoning
        - accentdb
        - aeslc
        - aflw2k3d
        - ag_news_subset
        - ai2_arc
        - ai2_arc_with_ir
        - amazon_us_reviews
        - anli
        - answer_equivalence
        - arc
        - asqa
        - asset
        - assin2
        - bair_robot_pushing_small
        - bccd
        - beans
        - bee_dataset
        - beir
        - big_patent
        - bigearthnet
        - billsum
        - binarized_mnist
        - binary_alpha_digits
        - ble_wind_field
        - blimp
        - booksum
        - bool_q
        - bot_adversarial_dialogue
        - bucc
        - c4
        - c4_wsrs
        - caltech101
        - caltech_birds2010
        - caltech_birds2011
        - cardiotox
        - cars196
[... 313 lines cut ...]
        - wsc273
        - xnli
        - xquad
        - xsum
        - xtreme_pawsx
        - xtreme_pos
        - xtreme_s
        - xtreme_xnli
        - yahoo_ltrc
        - yelp_polarity_reviews
        - yes_no
        - youtube_vis

Check that:
    - if dataset was added recently, it may only be available
      in `tfds-nightly`
    - the dataset name is spelled correctly
    - dataset class defines all base class abstract methods
    - the module defining the dataset class is imported

The builder directory /cluster/data7a/whq/data/bridge_orig doesn't contain any versions.
No builder could be found in the directory: /cluster/data7a/whq/data for the builder: bridge_orig.
No registered data_dirs were found in:
        - /cluster/data7a/whq/data
```

kpertsch, 2024-09-02:
It looks like this is an issue with the `bridge_orig` dataset you have stored. Here is what the expected folder structure should look like:
```
/cluster/data7a/whq/data/bridge_orig
          |
          -- 1.0.0
                 | -- features.json
                 | -- dataset_info.json
                 | -- bridge_dataset-train.tfrecord-00000-of-01024
                 | -- bridge_dataset-train.tfrecord-00001-of-01024
                 ...
```
Can you please verify that your folder structure looks like this

closed 2024-09-12 by kpertsch

split600, 2024-10-12:
> 我不能 100% 确定这个错误来自哪里——您能发布一个更长的错误堆栈跟踪吗？
> 
> 作为第一步调试，您可以尝试通过简单运行来加载数据集：
> 
> ```
> import tensorflow_datasets as tfds
> ds = tfds.load("bridge_orig", data_dir="/cluster/data7a/whq/data", split="train")
> ```
> 
> 如果出现错误，则原始数据可能存在问题。另外要确保：我假设 bridge_orig/1.9.0 目录中还有更多文件，对吗？似乎应该有 1024 个分片文件和`dataset_info.json`一个名为的文件`features.json`。
> 
> 附言：最好不要在这种公开可见的论坛上发布您的 huggingface 令牌，这会允许其他人访问您的帐户。

so if i run the code you provide above   ( import tensorflow_datasets as tfds
> ds = tfds.load("bridge_orig", data_dir="/cluster/data7a/whq/data", split="train")) . it still have the same bug  , does it mean  something wrong about the datasets

ClclEl, 2024-10-17:
Hello, I have the same problem, have u figure it out? :)

weichow23, 2024-12-31:
Maybe This will be helpful [LIBERO_modified_RLDS problem](https://github.com/openvla/openvla/issues/158)

LukeLIN-web, 2025-03-25:
> It looks like this is an issue with the `bridge_orig` dataset you have stored. Here is what the expected folder structure should look like:
>
> ```
> /cluster/data7a/whq/data/bridge_orig [...]

Thank you, I create a data/ dir and put bridge_orig inside data/ dir , it works.

LukeLIN-web, 2025-03-25:
I also meet `tensorflow.python.framework.errors_impl.InvalidArgumentError: {{function_node __wrapped__IteratorGetNext_output_types_22_device_/job:localhost/replica:0/task:0/device:CPU:0}} Requires start <= limit when delta > 0: 0/-2` Bug, and I decrease shuffle buffer to 90k . It is solved.