openvla/openvla #87: unnorm_key

RussRobin, 2024-09-04:
Hi

Thanks for the OpenVLA! I followed your instructions to finetune the model with lora on my own dataset, and tried to test model output with quickstart codes, where i change the path to the local ckpt path. I wonder how should i set unnorm_key?|

The thrown error:
```
  File "/root/.cache/huggingface/modules/transformers_modules/OpenVLA-hf-ckpt+MY_DATASET_NAME+b16+lr-0.0005+lora-r32+dropout-0.0--image_aug/modeling_prismatic.py", line 540, in _check_unnorm_key
    assert len(norm_stats) == 1, (
AssertionError: Your model was trained on more than one dataset, please pass a `unnorm_key` from the following options to choose the statistics used for un-normalizing actions: dict_keys(['austin_buds_dataset_converted_externally_to_rlds', 'austin_sailor_dataset_converted_externally_to_rlds', 'austin_sirius_dataset_converted_externally_to_rlds', 'bc_z', 'berkeley_autolab_ur5', 'berkeley_cable_routing', 'berkeley_fanuc_manipulation', 'bridge_orig', 'cmu_stretch', 'dlr_edan_shared_control_converted_externally_to_rlds', 'dobbe', 'fmb_dataset', 'fractal20220817_data', 'furniture_bench_dataset_converted_externally_to_rlds', 'iamlab_cmu_pickup_insert_converted_externally_to_rlds', 'jaco_play', 'kuka', 'nyu_franka_play_dataset_converted_externally_to_rlds', 'roboturk', 'stanford_hydra_dataset_converted_externally_to_rlds', 'taco_play', 'toto', 'ucsd_kitchen_dataset_converted_externally_to_rlds', 'utaustin_mutex', 'viola'])
```

All these keys are used in openvla pretraining, instead of the finetune dataset i use. So i do not know how to set unnorm_key.

Regards

moojink, 2024-09-05:
Hi @RussRobin,

Upon fine-tuning OpenVLA, a `dataset_statistics.json` file containing the `unnorm_key` and corresponding action un-normalization statistics should have been stored in your run directory. You should load this in and store the un-normalization statistics inside the VLA object before running inference. Please see an example of how to do this in the `openvla_utils.py` file > function `get_vla()` as shown [here](https://github.com/openvla/openvla/blob/abe0a22d91e8fb8a3c2930f326309a636c26bc1d/experiments/robot/openvla_utils.py#L59).

Also, we recently pushed another inference example -- this time in LIBERO simulation so that people can try OpenVLA even if they don't have physical robot hardware. Please see the [LIBERO Simulation Benchmark Evaluations](https://github.com/openvla/openvla?tab=readme-ov-file#libero-simulation-benchmark-evaluations) section in the updated README for details. In particular, you may find our LIBERO eval script ([run_libero_eval.py](https://github.com/openvla/openvla/blob/main/experiments/robot/libero/run_libero_eval.py)) useful to consult when trying to run fine-tuned OpenVLA models for your own use case.

-Moo Jin

RussRobin, 2024-09-05:
Thanks for your reply!

closed 2024-09-05 by RussRobin

Edingburge, 2025-06-25:
> Hi [@RussRobin](https://github.com/RussRobin),
>
> Upon fine-tuning OpenVLA, a `dataset_statistics.json` file containing the `unnorm_key` and corresponding action un-normalization statistics should [...]

Thank you. This method works very well. This method is applicable to custom-built datasets, where you load the dataset_statistics and assign it to norm_stats.

**dataset_statistics_path = "./openvla-7b-finetuned-libero-all/dataset_statistics.json"
                if os.path.isfile(dataset_statistics_path):
                    with open(dataset_statistics_path, "r") as f:
                        norm_stats = json.load(f)
                    vla.norm_stats = norm_stats**