openvla/openvla #88: Support for Joint ActionEncoding

Dash83, 2024-09-04:
Is there a particular reason why, per this code snippet from `prismatic/vla/datasets/rlds/oxe/materialize.py`, `ActionEncoding.JOINT_POS` are not currently supported?

```python
    if dataset_kwargs["action_encoding"] not in [ActionEncoding.EEF_POS, ActionEncoding.EEF_R6]:
        raise ValueError(f"Cannot load `{dataset_name}`; only EEF_POS & EEF_R6 actions supported!")
```

Is this because none of the datasets in the OXE mixture used for pre-training OpenVLA use `ActionEncoding.JOINT_POS`, or is there a deeper technical reason?

Thanks!

moojink, 2024-09-05:
Hi @Dash83,

Yes, you are correct: We use relative end-effector position control for all datasets used for OpenVLA's OpenX pretraining. This is because we want to have a unified action space across all datasets to facilitate better transfer between them.

-Moo Jin

Dash83, 2024-09-05:
Hey @moojink,

Thanks for your answer. I figured that would be the case and I understand the motivation. Unfortunately, I feel that design decision is in tension with supporting more embodiments like mobile manipulators, multi-arm setups, or humanoids. If I'm not mistaken, the `action_tokenizer` already supports output to action spaces of different dimensions, correct? If there was support for `ActionEncoding.JOINT_POS` as well, we could finetune the model on all these different embodiments.

Other than potential issues with transfer learning between tasks and embodiment, do you foresee any other technical hurdles I should be aware of If I remove the aforementioned constraint from the code and finetune on a dataset with `ActionEncoding.JOINT_POS`?

Thanks again for your support
- Marco

kpertsch, 2024-09-12:
I think this is a reasonable thing to try, I don't see a fundamental issue why this shouldn't work! The only thing to be careful with: if you have two datasets that "look" very similar but use different action spaces, you may want to add a token to the prompt of the model that tells it which action space you are expecting it to predict. If all datasets are clearly visually different (as is the case for OpenX at the moment), the model can implicitly infer it from the input image.

closed 2024-09-12 by kpertsch