openvla/openvla #93: About LIBERO configs and transforms

object814, 2024-09-06:
Hi! Thanks for your work.
I am actually doing finetuning on LIBERO dataset as well but the result was not as good as yours even with your modified RLDS dataset. I  think it should be the difference between the definition for LIBERO dataset in
`openvla/prismatic/vla/datasets/rlds/oxe/configs.py` and `openvla/prismatic/vla/datasets/rlds/oxe/transforms.py`
can you kindly provide the corresponding code for LIBERO dataset in these files? Or probably you should consider pushing the modified files for LIBERO onto the repo so that everyone can try to finetun themselves.
Thanks a lot.

moojink, 2024-09-07:
Hi @object814,

Sure, we can provide that! We just pushed the configs/transforms to the main branch. See the changes here: https://github.com/openvla/openvla/commit/7a359de9bb8be6fd3c232d189784920dbc93391c

This is only lightly tested since I am traveling now, so please let me know if you run into any issues such as runtime errors!

-Moo Jin

object814, 2024-09-07:
Great thanks a lot!

closed 2024-09-07 by object814

object814, 2024-09-07:
Hi I encountered some error after I changed the config and transform, which was raised at:
```
trajectory["observation"]["EEF_state"] = trajectory["observation"]["state"][:, :6]
trajectory["observation"]["gripper_state"] = trajectory["observation"]["state"][:, -2:]  # 2D gripper state
```
inside `transforms.py`
I printed out the RLDS dataset you provided for LIBERO and didn't found "state" key under rlds_data["observation"], the codebase is quite complex and to be honest I did quite a few changes here and there so I am not 100% sure if this is caused by me or is that we are missing some changes for LIBERO
Nonetheless my current solution is just to comment the two lines above inside `transforms.py`, and it did not raise any further probelm because I believe inside RLDS dataset handling it actually did not use ['state'] afterall.

Again, this is just a reminder, if the fresh codebase does not encounter this problem, its perfect.