openvla/openvla #94: Installation Issue

LemonZhong, 2024-09-07:
(openvla) wy@huang-ESC8000-G4:/home/data/wy/znz/openvla$ pip install "flash-attn==2.5.5" --no-build-isolation
Collecting flash-attn==2.5.5
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fa5ccea5c90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /packages/3b/ff/33a4ccc778c56d2ab9845b5ea09ab15d368ac39fa945f0bd2bd2724e18e8/flash_attn-2.5.5.tar.gz
  Downloading flash_attn-2.5.5.tar.gz (2.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.5/2.5 MB 3.6 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: torch in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from flash-attn==2.5.5) (2.2.0)
Requirement already satisfied: einops in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from flash-attn==2.5.5) (0.8.0)
Requirement already satisfied: packaging in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from flash-attn==2.5.5) (24.1)
Requirement already satisfied: ninja in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from flash-attn==2.5.5) (1.11.1.1)
Requirement already satisfied: filelock in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (3.13.1)
Requirement already satisfied: typing-extensions>=4.8.0 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (4.11.0)
Requirement already satisfied: sympy in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (1.13.2)
Requirement already satisfied: networkx in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (3.2.1)
Requirement already satisfied: jinja2 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (3.1.4)
Requirement already satisfied: fsspec in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (2024.9.0)
Requirement already satisfied: nvidia-cuda-nvrtc-cu12==12.1.105 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.105)
Requirement already satisfied: nvidia-cuda-runtime-cu12==12.1.105 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.105)
Requirement already satisfied: nvidia-cuda-cupti-cu12==12.1.105 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.105)
Requirement already satisfied: nvidia-cudnn-cu12==8.9.2.26 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (8.9.2.26)
Requirement already satisfied: nvidia-cublas-cu12==12.1.3.1 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.3.1)
Requirement already satisfied: nvidia-cufft-cu12==11.0.2.54 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (11.0.2.54)
Requirement already satisfied: nvidia-curand-cu12==10.3.2.106 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (10.3.2.106)
Requirement already satisfied: nvidia-cusolver-cu12==11.4.5.107 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (11.4.5.107)
Requirement already satisfied: nvidia-cusparse-cu12==12.1.0.106 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.0.106)
Requirement already satisfied: nvidia-nccl-cu12==2.19.3 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (2.19.3)
Requirement already satisfied: nvidia-nvtx-cu12==12.1.105 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (12.1.105)
Requirement already satisfied: triton==2.2.0 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from torch->flash-attn==2.5.5) (2.2.0)
Requirement already satisfied: nvidia-nvjitlink-cu12 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from nvidia-cusolver-cu12==11.4.5.107->torch->flash-attn==2.5.5) (12.6.68)
Requirement already satisfied: MarkupSafe>=2.0 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from jinja2->torch->flash-attn==2.5.5) (2.1.3)
Requirement already satisfied: mpmath<1.4,>=1.1.0 in /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages (from sympy->torch->flash-attn==2.5.5) (1.3.0)
Building wheels for collected packages: flash-attn
  Building wheel for flash-attn (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      fatal: not a git repository (or any of the parent directories): .git
      
      
      torch.__version__  = 2.2.0+cu121
      
      
      /home/wy/miniconda3/envs/openvla/lib/python3.10/site-packages/setuptools/__init__.py:85: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
      !!
      
              ********************************************************************************
              Requirements should be satisfied by a PEP 517 installer.
              If you are using pip, you can try `pip install --use-pep517`.
              ********************************************************************************
      
      !!
        dist.fetch_build_eggs(dist.setup_requires)
      running bdist_wheel
      Guessing wheel URL:  https://github.com/Dao-AILab/flash-attention/releases/download/v2.5.5/flash_attn-2.5.5+cu122torch2.2cxx11abiFALSE-cp310-cp310-linux_x86_64.whl
      error: <urlopen error [Errno 110] Connection timed out>
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for flash-attn
  Running setup.py clean for flash-attn
Failed to build flash-attn
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (flash-attn)

closed 2024-09-07 by LemonZhong

wu-yutong-525, 2024-09-18:
Hi Lemon, may I kindly ask how do you solve this issue? I encountered something similar:

```
Collecting flash-attn==2.5.5
  Using cached flash_attn-2.5.5.tar.gz (2.5 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      fatal: not a git repository (or any of the parent directories): .git
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-glszguso/flash-attn_d32e8fb69bcd4f8d8e4f823ad4328d86/setup.py", line 114, in <module>
          raise RuntimeError(
      RuntimeError: FlashAttention is only supported on CUDA 11.6 and above.  Note: make sure nvcc has a supported version by running nvcc -V.
      
      
      torch.__version__  = 2.2.0+cu121
      
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

```

kpertsch, 2024-09-18:
Your error indicates that your CUDA version does not support flash-attention. You may want to update CUDA first before installing flash attention.

billamiable, 2025-01-21:
Need at least CUDA 12