openvla/openvla #97: Generating `libero_spatial_no_noops`, `libero_object_no_noops`, `libero_goal_no_noops`, `libero_10_no_noops`.

branyang02, 2024-09-09:
Will the code for generating `libero_spatial_no_noops`, `libero_object_no_noops`, `libero_goal_no_noops`, `libero_10_no_noops` be released?

Converting LIBERO's hdf5 dataset to RLDS format is not a trivial task, and if the authors of this codebase could release the source code to generate the RLDS format of LIBERO's dataset that would be greatly appreciated!

kpertsch, 2024-09-14:
Hey, Moo Jin is currently travelling, but to unblock your research, here's a LIBERO --> RLDS converter: https://github.com/kpertsch/libero_dataset_builder

Note that this is not the one used for our LIBERO experiments since it doesn't contain the 0-action filtering -- once Moo Jin is back he can post his converter, but in the meantime this gives you an idea how to convert LIBERO to RLDS.

moojink, 2024-09-17:
Hi @branyang02,

We just pushed a commit that adds the script we used to regenerate the LIBERO datasets with our modifications, in the normal LIBERO HDF5 file format, and updated the README with a pointer to the HDF5-to-RLDS conversion script. See the commit here for details: [https://github.com/openvla/openvla/commit/0ebc5e333ee7a916813bf50db240f1af7ba1d295](https://github.com/openvla/openvla/commit/0ebc5e333ee7a916813bf50db240f1af7ba1d295)

Hope that helps!

-Moo Jin

closed 2024-09-18 by kpertsch