openvla/openvla #99: Fine tune pipeline for LIBERO

jbonobono, 2024-09-09:
Hello,
thank you for your great work openvla and recent update for LIBERO fine-tune model and eval pipeline

if possible, would you please share also fine tune pipeline for LIBERO? which can re-generate uploaded  LIBERO fine-tune model.

since I  want to do more experiment on LIBERO with backbone openvla, 
wanted to start same baseline as yours. if possible.

hope I can have your feedback

have a nice day
thank you

kpertsch, 2024-09-12:
We registered the LIBERO dataset in our codebase: https://github.com/openvla/openvla/blob/7a359de9bb8be6fd3c232d189784920dbc93391c/prismatic/vla/datasets/rlds/oxe/mixtures.py#L196

So you can download them from huggingface (https://huggingface.co/datasets/openvla/modified_libero_rlds) and then just run our default finetuning script, just make sure to set `--dataset_name` to the libero dataset you want to use!

closed 2024-09-12 by kpertsch

jbonobono, 2024-09-23:
thanks a lot !

jis3613, 2024-09-23:
Hi, I'm trying to finetune the openvla on libero dataset you provided.

However, I constantly get the following error.
`tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset libero_spatial_no_noops not found`
It seems like the libero dataset you modified in rlds format is not registered in the tfds.

Or is there additional steps that I need to follow ?

Your answer would be a great help to me.
Thanks!