openvla/openvla #101: OpenVLA Training Dataset Mixture

DelinQu, 2024-09-11:
Hi, 
thanks so much for sharing the awesome project. I'm trying to start the training with `Open-X Magic Soup++` data mixture configuration. The training logs show sample weights after preparing the dataset:
https://github.com/openvla/openvla/blob/7a359de9bb8be6fd3c232d189784920dbc93391c/prismatic/vla/datasets/rlds/dataset.py#L457
 
[image]

The real logging weights are different from the original paper.

[image]

So I carefully check the dataset version, tfrecord file length, and make sure my downloading is precisely correct. Besides, I calculate the ratio of `logging weights / original paper weights`:
```python
ours={
    "fractal20220817_data": 0.104098,
    "kuka": 0.363943,
    "bridge_orig": 0.108547,
    "taco_play": 0.024175,
    "jaco_play": 0.003963,
    "berkeley_cable_routing": 0.002152,
    "roboturk": 0.019062,
    "viola": 0.007759,
    "berkeley_autolab_ur5": 0.009957,
    "toto": 0.016555,
    "language_table": 0.035812,
    "stanford_hydra_dataset_converted_externally_to_rlds": 0.036418,
    "austin_buds_dataset_converted_externally_to_rlds": 0.001734,
    "nyu_franka_play_dataset_converted_externally_to_rlds": 0.006843,
    "furniture_bench_dataset_converted_externally_to_rlds": 0.020068,
    "ucsd_kitchen_dataset_converted_externally_to_rlds": 0.000404,
    "austin_sailor_dataset_converted_externally_to_rlds": 0.017948,
    "austin_sirius_dataset_converted_externally_to_rlds": 0.014229,
    "dlr_edan_shared_control_converted_externally_to_rlds": 0.000454,
    "iamlab_cmu_pickup_insert_converted_externally_to_rlds": 0.007433,
    "utaustin_mutex": 0.018395,
    "berkeley_fanuc_manipulation": 0.006365,
    "cmu_stretch": 0.001272,
    "bc_z": 0.061154,
    "fmb_dataset": 0.017190,
    "dobbe": 0.011588,
    "droid": 0.082481,
}

openvla={
    "fractal20220817_data": 12.7,
    "kuka": 12.7,
    "bridge_orig": 13.3,
    "taco_play": 3.0,
    "jaco_play": 0.4,
    "berkeley_cable_routing": 0.2,
    "roboturk": 2.3,
    "viola": 0.9,
    "berkeley_autolab_ur5": 1.2,
    "toto": 2.0,
    "language_table": 4.4,
    "stanford_hydra_dataset_converted_externally_to_rlds": 4.4,
    "austin_buds_dataset_converted_externally_to_rlds": 0.2,
    "nyu_franka_play_dataset_converted_externally_to_rlds": 0.8,
    "furniture_bench_dataset_converted_externally_to_rlds": 2.4,
    "ucsd_kitchen_dataset_converted_externally_to_rlds": 0.1,
    "austin_sailor_dataset_converted_externally_to_rlds": 2.2,
    "austin_sirius_dataset_converted_externally_to_rlds": 1.7,
    "dlr_edan_shared_control_converted_externally_to_rlds": 0.1,
    "iamlab_cmu_pickup_insert_converted_externally_to_rlds": 0.9,
    "utaustin_mutex": 2.2,
    "berkeley_fanuc_manipulation": 0.7,
    "cmu_stretch": 0.2,
    "bc_z": 7.5,
    "fmb_dataset": 7.1,
    "dobbe": 1.4,
    "droid": 10.0,   
}

for k, v in ours.items():
    print(f"{k:60s} {openvla[k]*0.01:.08f} -> {v:.08f}, ratio {v / openvla[k] * 100:.08f}")
```
The `logging weights / original paper weights` show that most datasets are correct with approximate ratio of 0.8, but the `kuka` and `fmb_dataset` are quite distinct. This indicates that the `kuka` and `fmb_dataset` might have undergone some changes, although my `version`, `raw data`, and even the `file sizes` are consistent with those in openvla and [openx](https://docs.google.com/spreadsheets/d/1rPBD77tk60AEIGZrGSODwyyzs5FgCU9Uz3h-3_t2A9g/edit?gid=0#gid=0). Could you confirm if the datasets used in your training are up to date, or if I might be missing any key details? I look forward to your feedback 🤗.
```
fractal20220817_data                                         0.12700000 -> 0.10409800, ratio 0.81966929
kuka                                                         0.12700000 -> 0.36394300, ratio 2.86569291
bridge_orig                                                  0.13300000 -> 0.10854700, ratio 0.81614286
taco_play                                                    0.03000000 -> 0.02417500, ratio 0.80583333
jaco_play                                                    0.00400000 -> 0.00396300, ratio 0.99075000
berkeley_cable_routing                                       0.00200000 -> 0.00215200, ratio 1.07600000
roboturk                                                     0.02300000 -> 0.01906200, ratio 0.82878261
viola                                                        0.00900000 -> 0.00775900, ratio 0.86211111
berkeley_autolab_ur5                                         0.01200000 -> 0.00995700, ratio 0.82975000
toto                                                         0.02000000 -> 0.01655500, ratio 0.82775000
language_table                                               0.04400000 -> 0.03581200, ratio 0.81390909
stanford_hydra_dataset_converted_externally_to_rlds          0.04400000 -> 0.03641800, ratio 0.82768182
austin_buds_dataset_converted_externally_to_rlds             0.00200000 -> 0.00173400, ratio 0.86700000
nyu_franka_play_dataset_converted_externally_to_rlds         0.00800000 -> 0.00684300, ratio 0.85537500
furniture_bench_dataset_converted_externally_to_rlds         0.02400000 -> 0.02006800, ratio 0.83616667
ucsd_kitchen_dataset_converted_externally_to_rlds            0.00100000 -> 0.00040400, ratio 0.40400000
austin_sailor_dataset_converted_externally_to_rlds           0.02200000 -> 0.01794800, ratio 0.81581818
austin_sirius_dataset_converted_externally_to_rlds           0.01700000 -> 0.01422900, ratio 0.83700000
dlr_edan_shared_control_converted_externally_to_rlds         0.00100000 -> 0.00045400, ratio 0.45400000
iamlab_cmu_pickup_insert_converted_externally_to_rlds        0.00900000 -> 0.00743300, ratio 0.82588889
utaustin_mutex                                               0.02200000 -> 0.01839500, ratio 0.83613636
berkeley_fanuc_manipulation                                  0.00700000 -> 0.00636500, ratio 0.90928571
cmu_stretch                                                  0.00200000 -> 0.00127200, ratio 0.63600000
bc_z                                                         0.07500000 -> 0.06115400, ratio 0.81538667
fmb_dataset                                                  0.07100000 -> 0.01719000, ratio 0.24211268
dobbe                                                        0.01400000 -> 0.01158800, ratio 0.82771429
droid                                                        0.10000000 -> 0.08248100, ratio 0.82481000
```

kpertsch, 2024-09-12:
For Kuka, you need to run success filtering after downloading the raw dataset -- if you check the script here it runs the success filter: https://github.com/moojink/rlds_dataset_mod/blob/ad83e6c0efad5823540c0f6d3a05529596ead0b5/prepare_open_x.sh#L33

For FMB, there has been an issue where only a partial dataset was uploaded to the OpenX GCP bucket and it has been in processing for a long time...
If this is blocking you, feel free to shoot me an email and I can share the FMB dataset before it gets fixed on the google bucket... (karl.pertsch@gmail.com)

closed 2024-09-12 by kpertsch

DelinQu, 2024-09-16:
Thanks [kpertsch](https://github.com/kpertsch),  I have sent an email request. 😃

Shinechaote, 2024-09-18:
@kpertsch Do you know whether the FMB dataset has been fixed?

Lucky-Light-Sun, 2025-02-12:
Hello, have you ever fixed this problem?

I run kuka download script in [https://github.com/moojink/rlds_dataset_mod/blob/ad83e6c0efad5823540c0f6d3a05529596ead0b5/prepare_open_x.sh#L33](https://github.com/moojink/rlds_dataset_mod/blob/ad83e6c0efad5823540c0f6d3a05529596ead0b5/prepare_open_x.sh#L33). And I think i may run success filtering after downloading the raw dataset.  So what's the reason that causes kuka weight problem?

What's more, after downloading all FMB, are dataset training weights consistent with those in the paper?

Looking forward to your reply!