openvla/openvla #104: Discrepancy in training vs inference prompt

oceanusxiv, 2024-09-13:
Hi, I noticed that in the base prompter used for training, https://github.com/openvla/openvla/blob/7a359de9bb8be6fd3c232d189784920dbc93391c/prismatic/models/backbones/llm/prompting/base_prompter.py#L36
there is a space at the end of `Out`, however, all the examples for inference, such as https://github.com/openvla/openvla/blob/7a359de9bb8be6fd3c232d189784920dbc93391c/experiments/robot/openvla_utils.py#L163
seem to remove that space, what's the reasoning behind this discrepancy? Shouldn't we want the prompt to be identical between training and inference?

moojink, 2024-09-16:
Hi @oceanusxiv,

Thank you for the question!

**TLDR:** It turns out everything works fine and there is no discrepancy between train time and test time if you run the inference code as is, because the OpenVLA `predict_action()` function adds a special empty token to the end of the prompt, before the action tokens are predicted, to resolve the discrepancy.

----

**Detailed explanation:**

Consider an example where the task is to `"pick up the black bowl next to the ramekin and place it on the plate"`. Then, the prompt and response at train time would be the following (excluding the image for simplicity):

`"<s>In: What action should the robot take to pick up the black bowl next to the ramekin and place it on the plate?\nOut: ABCDEFG</s>"`,

where `ABCDEFG` denote the 7 robot action tokens OpenVLA needs to predict. (These are just symbols -- not the literal tokens. The real tokens might look something like `貴학 ਿ˚陈 ਿ忠`, which come from the set of last 256 tokens in the vocabulary.)

OpenVLA's tokenization of the above produces the following `input_ids`:

`[1, 512, 29901, 1724, 3158, 881, 278, 19964, 2125, 304, 5839, 701, 278, 4628, 12580, 29880, 2446, 304, 278, 364, 420, 9089, 322, 2058, 372, 373, 278, 15284, 29973, 13, 3744, 29901, 29871, 31887, 31822, 31857, 31878, 31862, 31857, 31744, 2]`

You can use the tokenizer to decode the `input_ids` into individual tokens like so:

```
id: 1           token: <s>
id: 512         token: In
id: 29901       token: :
id: 1724        token: What
id: 3158        token: action
id: 881         token: should
id: 278         token: the
id: 19964       token: robot
id: 2125        token: take
id: 304         token: to
id: 5839        token: pick
id: 701         token: up
id: 278         token: the
id: 4628        token: black
id: 12580       token: bow
id: 29880       token: l
id: 2446        token: next
id: 304         token: to
id: 278         token: the
id: 364         token: r
id: 420         token: ame
id: 9089        token: kin
id: 322         token: and
id: 2058        token: place
id: 372         token: it
id: 373         token: on
id: 278         token: the
id: 15284       token: plate
id: 29973       token: ?
id: 13          token: <NEWLINE>
id: 3744        token: Out
id: 29901       token: :
id: 29871       token: 
id: 31887       token: 貴
id: 31822       token: 학
id: 31857       token: ਿ
id: 31878       token: ˚
id: 31862       token: 陈
id: 31857       token: ਿ
id: 31744       token: 忠
id: 2           token: </s>
```

At test time, if you use the prompt `f"In: What action should the robot take to {task_label.lower()}?\nOut:"` with no space character at the end and then break it down into individual tokens like above, then you will see that the `input_ids` looks the same as above, except that the token with ID `29871` (representing the empty `''` token) is missing. At first glance, this may appear to be a problem, but the OpenVLA `predict_action()` function actually handles this by adding the empty token to the prompt, as shown [here](https://huggingface.co/openvla/openvla-7b/blob/main/modeling_prismatic.py#L513). Therefore, there is no discrepancy between train and test time.

If you change the prompt and add a space character (i.e., use `"Out: "` instead of `"Out:"`), that's still fine because `predict_action()` performs a check and doesn't add the empty token if it already exists at the end of the prompt, as shown [here](https://huggingface.co/openvla/openvla-7b/blob/main/modeling_prismatic.py#L512).
* **[Note (2024-09-16)]:** We added this check to the public GitHub repo some time in the past, as shown in [this commit](https://github.com/openvla/openvla/commit/d7dbea892e0772716f198c31fb1a1da24f4bb129), but recently realized that the change wasn't reflected in the Hugging Face OpenVLA inference code that is downloaded to cache along with the model checkpoint. Therefore, we just updated the Hugging Face code today to match the GitHub repo (you can clear your Hugging Face OpenVLA cache to see the changes locally). However, this should not have caused any issues if you used the inference code exactly as shown in `openvla_utils.py`.

Hope that helps clears things up!

-Moo Jin

oceanusxiv, 2024-09-16:
@moojink Thanks for the detailed response! Yeah I actually figured this out on my own and noticed this code that was added. A couple things though.
1. I don't think this code is added to the `prediction_action` of the `OpenVLA` class in `openvla.py`. The original pytorch model, not the converted huggingface one. So if you loaded the checkpoint via the original pytorch class and ran inference on it I think it will fail.
2. FWIW, and I know it's probably not relevant to the vast majority of users, but Mistral uses the same llama tokenizer, but has a different vocabulary config, so you'd end up with a different "hidden token" compared to a llama2 based model.

moojink, 2024-09-17:
Hi @oceanusxiv,

1. Thank you for pointing that out in `openvla.py`! We just pushed a [commit](https://github.com/openvla/openvla/commit/c40d1ffa6f673d08ef36dbaa123425e9638e0b7a) to address that.
2. Good point. At some point we did have code to handle cases like that but refactored it away since we didn't end up developing OpenVLA with different base LLMs and we didn't want the code to be too clunky or confusing to people. But yes, customizing the model like that will require tweaking the `predict_action()` function definition!

-Moo Jin

closed 2024-09-18 by kpertsch