openvla/openvla #106: LIBERO Problem

LemonZhong, 2024-09-16:
(openvla) wangyi@amax:/data/wy/znz/openvla$ python experiments/robot/libero/run_libero_eval.py \
>   --model_family openvla \
>   --pretrained_checkpoint openvla/openvla-7b-finetuned-libero-spatial \
>   --task_suite_name libero_spatial \
>   --center_crop True
2024-09-16 15:15:04.726175: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-09-16 15:15:04.777282: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-09-16 15:15:04.777331: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-09-16 15:15:04.778716: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-09-16 15:15:04.786655: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-09-16 15:15:05.705885: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/scripts/setup_macros.py (macros.py:55)
[*] Instantiating Pretrained VLA model
[*] Loading in BF16 with Flash-Attention Enabled
<frozen importlib._bootstrap>:283: DeprecationWarning: the load_module() method is deprecated and slated for removal in Python 3.12; use exec_module() instead
Loading checkpoint shards: 100%|███████████████████████████████████████████████████████████████████████████████████| 4/4 [00:01<00:00,  2.87it/s]
WARNING: No local dataset_statistics.json file found for current checkpoint.
You can ignore this if you are loading the base VLA (i.e. not fine-tuned) checkpoint.Otherwise, you may run into errors when trying to call `predict_action()` due to an absent `unnorm_key`.
Loaded model: <class 'transformers_modules.openvla.openvla-7b.e5822cc24559b04e532f49b1c1ddb64376c1a485.modeling_prismatic.OpenVLAForActionPrediction'>
Traceback (most recent call last):
  File "/amax/data/wy/znz/openvla/experiments/robot/libero/run_libero_eval.py", line 286, in <module>
    eval_libero()
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/amax/data/wy/znz/openvla/experiments/robot/libero/run_libero_eval.py", line 117, in eval_libero
    processor = get_processor(cfg)
  File "/amax/data/wy/znz/openvla/experiments/robot/openvla_utils.py", line 77, in get_processor
    processor = AutoProcessor.from_pretrained(cfg.pretrained_checkpoint, trust_remote_code=True)
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/transformers/models/auto/processing_auto.py", line 342, in from_pretrained
    raise ValueError(
ValueError: Unrecognized processing class in openvla/openvla-7b-finetuned-libero-spatial. Can't instantiate a processor, a tokenizer, an image processor or a feature extractor for this model. Make sure the repository contains the files of at least one of those processing classes.

moojink, 2024-09-17:
Hi @LemonZhong,

I have tried rerunning the command you pasted above but am not encountering any errors on my end. Can you double check that the fine-tuned OpenVLA checkpoint has been downloaded to your Transformers cache directory correctly? It should be about ~15 GB total and contain files such as `config.json`, `preprocessor_config.json`, `tokenizer_config.json`, etc. In addition, could you check that you have the latest version of the `main` branch?

-Moo Jin

LemonZhong, 2024-09-18:
I have solved the problem, thanks! Because I have used VPN to download the ckpt and transmit it to the remote server, I need to have a different location to load the ckpt. And now I have solved it.

closed 2024-09-18 by LemonZhong