openvla/openvla #108: Guidance needed for running LIBERO evaluation on headless server or in Docker environment

wlxing1901, 2024-09-16:
First, I want to express my sincere admiration for the OpenVLA project. Your work on creating an open-source Vision-Language-Action model is truly groundbreaking, and the comprehensive nature of your research greatly benefits the entire robotics and AI community.

While exploring your excellent project, I've successfully run inference on minimal examples, but I've encountered challenges when attempting to run the LIBERO evaluation on a headless server. I'm seeking guidance on how to properly use simulators for evaluation in **a headless server environment** or within **a Docker container**.

Specifically, I'm receiving an EGL-related error when trying to import modules necessary for the LIBERO evaluation. Here's the error message:

`Traceback (most recent call last):
  File "/home/xwl/Projects/openvla/experiments/robot/libero/run_libero_eval.py", line 35, in <module>
    from experiments.robot.libero.libero_utils import (
  File "/home/xwl/Projects/openvla/experiments/robot/libero/libero_utils.py", line 10, in <module>
    from libero.libero.envs import OffScreenRenderEnv
  File "/home/xwl/Projects/LIBERO/libero/libero/envs/__init__.py", line 1, in <module>
    from .bddl_base_domain import TASK_MAPPING
  File "/home/xwl/Projects/LIBERO/libero/libero/envs/bddl_base_domain.py", line 3, in <module>
    import robosuite.utils.transform_utils as T
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/utils/binding_utils.py", line 58, in <module>
    from robosuite.renderers.context.egl_context import EGLGLContext as GLContext
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/renderers/context/egl_context.py", line 32, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 31, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/mujoco/egl/egl_ext.py", line 27, in <module>
    from OpenGL import EGL
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/home/xwl/Libraries/miniconda3/envs/openvla/lib/python3.10/site-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
    raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
AttributeError: 'NoneType' object has no attribute 'eglQueryString'`

Shinechaote, 2024-09-16:
I have the same issue. By installing libegl-dev I was able to get to the following error message:

Traceback (most recent call last):
  File "/home/helloworld/openvla/experiments/robot/libero/run_libero_eval.py", line 35, in <module>
    from experiments.robot.libero.libero_utils import (
  File "/home/helloworld/openvla/experiments/robot/libero/libero_utils.py", line 10, in <module>
    from libero.libero.envs import OffScreenRenderEnv
  File "/home/helloworld/LIBERO/libero/libero/envs/__init__.py", line 1, in <module>
    from .bddl_base_domain import TASK_MAPPING
  File "/home/helloworld/LIBERO/libero/libero/envs/bddl_base_domain.py", line 3, in <module>
    import robosuite.utils.transform_utils as T
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/utils/binding_utils.py", line 58, in <module>
    from robosuite.renderers.context.egl_context import EGLGLContext as GLContext
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/robosuite/renderers/context/egl_context.py", line 32, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/helloworld/miniconda3/envs/openvla/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 67, in <module>
    raise ImportError(
ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.

Shinechaote, 2024-09-17:
Hey, I was able to solve the problem:
1. I installed libgl1-mesa-dri libgl1-mesa-dev libgl1-mesa-glx libstdc++6
2. I then set LIBGL_DRIVERS_PATH=/path/to/where/the/drivers/are (in my case /usr/lib/x86_64-linux-gnu/dri/)
3. Because I was using conda I had to link libstdc++.so.6 to my conda lib env 
(ln -sf /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /home/helloworld/miniconda3/envs/openvla/lib/libstdc++.so.6)

kpertsch, 2024-09-18:
Thanks for posting your solution!
If you have more questions that are LIBERO-specific, please post them in the original LIBERO repository!

closed 2024-09-18 by kpertsch

wlxing1901, 2024-09-18:
@Shinechaote Thank you for your help! The solution worked perfectly, and now everything is running smoothly. Really appreciate it!