openvla/openvla #109: Data Parallelism for RLDS Format IterableDataset:

jiayueru, 2024-09-17:
Thank you very much for your excellent work!
I noticed in finetune.py that if using the RLDS format, explicit parallelism settings are not required. However, when I run the code on a single machine with 8 GPUs, I observe that each GPU processes the same amount of data as a single GPU. For example, I want to process the Viola subset in Open-X, where dataset_length = 76324. When I set batch_size = 4, I expect to have 19081 * 4 batches, which should result in 19081 batches per GPU. However, I find that each GPU still processes 19081 batches. Could you please help me understand why this is happening?
[image]

kpertsch, 2024-09-18:
Note that our RLDS data loader calls `repeat()` on the dataset, so there's no strict notion of epochs and the data loader will just keep spinning through the dataset as you request new batches. So you will need to manually ensure that you stop training after samples / batch_size iterations if you only want to train for one epoch.

closed 2024-09-18 by kpertsch

jiayueru, 2024-09-18:
Thank you! If I only want to go through the dataset once, do I need to manually modify the iterable dataset to ensure that each GPU processes different data? As it stands, I cannot guarantee that each GPU will receive different data within a single loop.

kpertsch, 2024-09-18:
Yes, I think you'd need to modify the underlying RLDS data loader to not repeat data