openvla/openvla #110: On the RTX8000, finetune.py fails during execution even when set with parameters that consume less memory.

shifsa, 2024-09-17:
I attempted to run finetune.py on BridgeDataV2. Due to previous errors in the code, I changed bfloat16 to float32 and commented out attn_implementation. Despite setting the parameters to minimize memory consumption (--lora_rank 1 --batch_size 1 --grad_accumulation_steps 1), I still encountered the error shown below.
Is there any other way to reduce memory consumption? Or is it inherently difficult to finetune in this environment?

Environment:
- OS: Ubuntu20.04
- RAM: 64GB
- CPU: Intel(R) Core(TM) i9-10900X
- GPU: Quadro RTX 8000
- CUDA version: 12.0

```
(openvla) shimizke@linux:~/openvla$ torchrun --standalone --nnodes 1 --nproc-per-node 1 vla-scripts/finetune.py  --vla_path "openvla/openvla-7b"  --data_root_dir "test_dataset"  --dataset_name bridge_orig  --run_root_dir "checkpoints_log"  --adapter_tmp_dir "checkpoints_log/adapter_tmp"  --lora_rank 1  --batch_size 1  --grad_accumulation_steps 1  --learning_rate 5e-4  --image_aug False  --wandb_project xx  --wandb_entity xx  --save_steps 2 --max_steps 2
2024-09-17 12:51:12.271071: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-09-17 12:51:12.910081: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-09-17 12:51:12.910287: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-09-17 12:51:13.006465: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-09-17 12:51:13.185750: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
Fine-tuning OpenVLA Model `openvla/openvla-7b` on `bridge_orig`
Loading checkpoint shards:   0%|                                                                                                                                    | 0/3 [00:00<?, ?it/s]Loading checkpoint shards: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3/3 [00:32<00:00, 10.97s/it]
trainable params: 3,463,384 || all params: 7,544,700,568 || trainable%: 0.0459
2024-09-17 12:52:14.670779: I tensorflow/core/grappler/optimizers/data/replicate_on_split.cc:32] Running replicate on split optimization
09/17 [12:52:15] INFO     | >> [*] Loading existing dataset statistics from                                                                                              data_utils.py:208
                          test_dataset/bridge_orig/1.0.0/dataset_statistics_190b974b6ae7dd751370de88a9c06039c7a7cc8f379a729919e4f11c4b647945.json.                                        
2024-09-17 12:52:15.554584: I tensorflow/core/grappler/optimizers/data/replicate_on_split.cc:32] Running replicate on split optimization

######################################################################################
# Loading the following 1 datasets (incl. sampling weight):                         #
# bridge_orig: =============================================================1.000000 #
######################################################################################

09/17 [12:52:16] INFO     | >> [*] Threads per Dataset: [1]                                                                                                                 dataset.py:531
                 INFO     | >> [*] Reads per Dataset: [1]                                                                                                                   dataset.py:532
                 INFO     | >> [*] Constructing datasets...                                                                                                                 dataset.py:535
2024-09-17 12:52:16.499813: I tensorflow/core/grappler/optimizers/data/replicate_on_split.cc:32] Running replicate on split optimization
09/17 [12:52:17] INFO     | >> [*] Applying frame transforms on dataset...                                                                                                  dataset.py:575
                 INFO     | >> [*] Saved dataset statistics file at path checkpoints_log/openvla-7b+bridge_orig+b1+lr-0.0005+lora-r1+dropout-0.0/dataset_statistics.json data_utils.py:293
wandb: (1) Create a W&B account
wandb: (2) Use an existing W&B account
wandb: (3) Don't visualize my results
wandb: Enter your choice: 3
wandb: WARNING Invalid choice
wandb: Enter your choice: 3 
wandb: You chose 'Don't visualize my results'
wandb: Tracking run with wandb version 0.13.1
wandb: W&B syncing is set to `offline` in this directory.  
wandb: Run `wandb online` or set WANDB_MODE=online to enable cloud syncing.
  0%|                                                                                                                                                               | 0/2 [00:00<?, ?it/s]2024-09-17 12:53:00.466610: I tensorflow/core/kernels/data/shuffle_dataset_op.cc:422] ShuffleDatasetV3:43: Filling up shuffle buffer (this may take a while): 54124 of 100000
2024-09-17 12:53:08.615404: I tensorflow/core/kernels/data/shuffle_dataset_op.cc:452] Shuffle buffer filled.
3it [00:27,  6.42s/it]                                                                                                                                                                    Saving Model Checkpoint for Step 2
/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/peft/utils/save_and_load.py:180: UserWarning: Setting `save_embedding_layers` to `True` as embedding layers found in `target_modules`.
  warnings.warn("Setting `save_embedding_layers` to `True` as embedding layers found in `target_modules`.")
Loading checkpoint shards: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3/3 [00:10<00:00,  3.66s/it]
Saved Model Checkpoint for Step 2 at: checkpoints_log/openvla-7b+bridge_orig+b1+lr-0.0005+lora-r1+dropout-0.0███████████████████████████████████████████████| 3/3 [00:10<00:00,  3.76s/it]
5it [01:52, 24.11s/it]Saving Model Checkpoint for Step 4
/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/peft/utils/save_and_load.py:180: UserWarning: Setting `save_embedding_layers` to `True` as embedding layers found in `target_modules`.
  warnings.warn("Setting `save_embedding_layers` to `True` as embedding layers found in `target_modules`.")
                                                                                                                                                                                         [2024-09-17 12:57:47,877] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: -9) local_rank: 0 (pid: 2693110) of binary: /home/shimizke/miniconda3/envs/openvla/bin/python
Traceback (most recent call last):
  File "/home/shimizke/miniconda3/envs/openvla/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 347, in wrapper
    return f(*args, **kwargs)
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 812, in main
    run(args)
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 135, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
[... 11 lines cut ...]
  host      : linux
  rank      : 0 (local_rank: 0)
  exitcode  : -9 (pid: 2693110)
  error_file: <N/A>
[... cut ...]
    msg = self._response_queue.get(timeout=1)
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/multiprocessing/queues.py", line 117, in get
    res = self._recv_bytes()
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/multiprocessing/connection.py", line 212, in recv_bytes
    self._check_closed()
  File "/home/shimizke/miniconda3/envs/openvla/lib/python3.10/multiprocessing/connection.py", line 136, in _check_closed
    raise OSError("handle is closed")
OSError: handle is closed

```

kpertsch, 2024-09-18:
mmh I think your GPU has 48GB of memory so I would expect LoRA finetuning to work on it. Could you try putting bfloat16 back in? It should reduce memory usage. 
If nothing works QLoRA would be the other way to reduce memory, but it hasn't trained stably for me in the past. LoRA should really just work with the size of your GPU.

shifsa, 2024-09-18:
I reverted everything back to bfloat16 but encountered the same error as before(#103). When I changed the error point to float32, an out-of-memory error occurred, so I have converted all instances of bfloat16 in the code to float32.

This time, I changed only the bfloat16 within torch.autocast("cuda", dtype=torch.bfloat16): to float16. This adjustment allowed the process to complete fine-tuning without any memory errors.

closed 2024-09-18 by shifsa

teamROBI, 2025-03-28:
Hello. I am using the same gpu and I have some questions.

RTX Q8000 hardware does not support bfloat16, so you changed to float16 to finetune right? Are there no losses of information and deploys well when testing? Because float16 represents smaller range than bfloat16.

Thank you.