openvla/openvla #112: Details of the finetuning settings for the LIBERO task

jiayueru, 2024-09-19:
In the "Launching LIBERO Evaluations" section, could you please provide the details about the finetuning process for the four tasks on the LIBERO task? For example, the number of epochs, batch size, iteration steps, etc.? The success rate of our self-finetuned results has been very low. 😭

Talnex, 2024-09-19:
I saw that the settings of batch size and iteration steps are explained in the readme of the official ckpt website link, and the other settings are from the default script of LoRA finetuning.

- [OpenVLA 7B Fine-Tuned on LIBERO-Spatial](https://huggingface.co/openvla/openvla-7b-finetuned-libero-spatial)
We fine-tuned OpenVLA with batch size 128 for 50K gradient steps using 8 A100 GPUs. 

- [OpenVLA 7B Fine-Tuned on LIBERO-Object](https://huggingface.co/openvla/openvla-7b-finetuned-libero-object)
We fine-tuned OpenVLA with batch size 128 for 50K gradient steps using 8 A100 GPUs. 

- [OpenVLA 7B Fine-Tuned on LIBERO-Goal](https://huggingface.co/openvla/openvla-7b-finetuned-libero-goal)
We fine-tuned OpenVLA with batch size 128 for 60K gradient steps using 8 A100 GPUs. 

- [OpenVLA 7B Fine-Tuned on LIBERO-10 (LIBERO-Long)](https://huggingface.co/openvla/openvla-7b-finetuned-libero-10)
We fine-tuned OpenVLA with batch size 128 for 80K gradient steps using 8 A100 GPUs. 

- Both
We applied random crop and color jitter image augmentations during training (therefore, center cropping should be applied at inference time.

jiayueru, 2024-09-19:
soso thx!

zhaoyu775885, 2024-09-20:
Have you reproduced the LIBERO simulation experiment? I set the batch size to be 32, and keep the default settings in fintune.py. With 50K updates, the accuracy was still below 50% for the LIBERO-goal task. 
Tried twice and not tested other tasks.

Talnex, 2024-09-20:
> Have you reproduced the LIBERO simulation experiment? I set the batch size to be 32, and kept the default settings in fintune.py. With 50K updates, the accuracy was still below 50% for the LIBERO-goal task. Tried twice and not tested other tasks.

I finetuned both the object and spatial datasets, but neither achieved the results in the paper. Here are my settings
- ckpts  from https://huggingface.co/openvla/openvla-7b/tree/main
- dataset from https://huggingface.co/datasets/openvla/modified_libero_rlds/tree/main
- 8 x A100(80G)
- batch_size=16 (16x8=128)
- learning_rate= 2e-5
- grad_accumulation_steps=1
- image_aug=True
- lora_rank=32
- lora_dropout=0.0
- use_quantization=False 
- Python 3.10.13, PyTorch 2.2.0, transformers 4.40.1, and flash-attn 2.5.5
- Tested using ckpt with step=60k

[image]
| Task            | Test Type          | Success Rate | Number of Trials |
|-----------------|--------------------|--------------|------------------|
| Object | Official ckpt   | 82.2%        | 101              |
| Object | My ckpt   | 5.4%         | 37               |
| Spatial | Official ckpt | 91.6%        | 178              |
| Spatial | My ckpt   | 50.8%        | 60               |

I noticed that most of the failures were caused by the robot arm not moving.

Talnex, 2024-09-22:
I think I found solution. My codebase is not updated to the latest version. The latest version of the code seems to have added processing for empty tokens.

rokas-bendikas, 2024-10-03:
I am also struggling to reproduce the results of finetuning on Libero dataset. My results are very comparable with @Talnex (same node setup and parameters used). Were there any additional modifications applied to the code/data that are not mentioned in the paper?

rokas-bendikas, 2024-10-07:
**UPDATE** Solved the plateau issue by increasing the learning rate to 1e-4. Would be interesting to know the official parameter used for the results reported in the paper.

moojink, 2024-10-09:
Hi everyone (@jiayueru @Talnex @zhaoyu775885 @rokas-bendikas),

Below are the hyperparameters we used for all LIBERO experiments:

- Hardware: 8 x A100 GPUs with 80GB memory, for every checkpoint
- Fine-tuned with LoRA: `use_lora == True`, `lora_rank == 32`, `lora_dropout == 0.0`
- Learning rate: 5e-4
- Batch size: 128 (8 GPUs x 16 samples each)
- Number of training gradient steps: 50K for LIBERO-Spatial, 50K for LIBERO-Object, 60K for LIBERO-Goal, 80K for LIBERO-10 (in general, we fine-tuned until action token accuracy stabilized at ~100% with image augmentations enabled)
- No quantization at train or test time
- No gradient accumulation (i.e. `grad_accumulation_steps == 1`)
- `shuffle_buffer_size == 100_000`

We'll update the model cards on Hugging Face to include these extra details!

-Moo Jin

closed 2024-10-15 by kpertsch

YEOMJINSEOP, 2024-11-11:
Hi @Talnex, i'm also experiencing similar issue. Did you solve this problem??

Talnex, 2024-11-29:
> Hi @Talnex, @rokas-bendikas, i'm also experiencing similar issue. Did you guys solve this issue??

[image]

I used full fine-tuning to reproduce the effect. The training accuracy can reach 98%+

| task    | ckpt          | ft setting     | test setting | success rate |
|---------|---------------|----------------|--------------|--------------|
| spatial | paper         | aug-2e-5-bs128 | aug-50K      | 84.7 ± 0.9%  | 
| spatial | official-lora | aug-2e-5-bs128 | aug-50K      | 83.4%        |
| spatial | full-ft       | aug-2e-5-bs128 | aug-10K      | 77.4%        | 

| task    | ckpt          | ft setting     | test setting | success rate |
|---------|---------------|----------------|--------------|--------------|
| object  | paper         | aug-2e-5-bs128 | aug-50K      | 88.4 ± 0.8%  | 
| object  | official-lora | aug-2e-5-bs128 | aug-50K      | 85.8%        | 
| object  | full-ft       | aug-2e-5-bs128 | aug-50K      | 85.6%        |

| task    | ckpt          | ft setting     | test setting | success rate |
|---------|---------------|----------------|--------------|--------------|
| goal    | paper         | aug-2e-5-bs128 | aug-60K      | 79.2 ± 1.0%  | 
| goal    | official-lora | aug-2e-5-bs128 | aug-60K      | 77.4%        | 
| goal    | full-ft       | aug-2e-5-bs128 | aug-30K      | 72.0%        |

| task    | ckpt          | ft setting     | test setting | success rate |
|---------|---------------|----------------|--------------|--------------|
| long    | paper         | aug-2e-5-bs128 | aug-80K      | 53.7 ± 1.3%  | 
| long    | official-lora | aug-2e-5-bs128 | aug-80K      | 54.0%        | 
| long    | full-ft       | aug-2e-5-bs128 | aug-80K      | 46.6%        | 

The ckpt of full fine-tuning is closer to the official ckpt result than Lora fineture. Goal  and object  task is comparable.
At the same time, I found that the ckpt I trained was not necessarily the best performance at the corresponding step.
I used osmesa as a virtual display for rendering LIBERO. :)

WalkerRusher, 2025-02-28:
@Talnex Hi,  I am wondering that how long have you taken for full fine-tuning the model? Could you please tell me your configuration details? (e.g., how many gpus used)?

Talnex, 2025-03-01:
> [@Talnex](https://github.com/Talnex) Hi, I am wondering that how long have you taken for full fine-tuning the model? Could you please tell me your configuration details? (e.g., how many gpus used)?

In order to search for the best ckpt, I trained twice the number of steps of the official ckpt. Each task takes about 8 x A100 training for 3.5 to 4 days. The training settings are in my reply above. They are consistent with the official ones. Although I have not fully reproduced all the accuracy of libero, some people I know claim to have achieved the official accuracy using only lora finetune.