openvla/openvla #113: Which prismatic?

tekntrash, 2024-09-19:
By just installing this repo, apparently it does not install prismatic as when one types train.py one gets ModuleNotFoundError: No module named 'prismatic'. As there is a prismatic folder in the repo, one would think it should import it from there?

If one then does pip install prismatic  one gets ModuleNotFoundError: No module named 'prismatic.conf'

So where is prismatic supposed to come from?

tekntrash, 2024-09-19:
I solved it by inserting the code 
import sys
sys.path.append('/root/openvla/')
import prismatic 

before the prismatic imports

Shinechaote, 2024-09-20:
You need to execute scripts/train.py from the openvla folder. 
As far as I can tell `pip install -e .` just installs the dependencies for running openvla but does not install prismatic as a package.
Therefore python cannot find prismatic except if you execute your script from the openvla folder.

siddk, 2024-09-23:
Yeah - you will need to either do an editable install of the repo as @Shinechaote mentions, or update your `PYTHONPATH` to include the repository root!

closed 2024-09-23 by siddk